roleoroleo/yi-hack-MStar #277: No video in ONVIF mode connected to Qnap Surveillance Station

Cho0Cho0, 2020-11-19:
Hi,
I'm using a 0.3.8 release for YI Dome X and in general it works great.

But I have a small issue with Qnap Surveillance Station (SS):
- adding camera as Generic RTSP works great
- camera can be found via ONVIF, but SS doesn't show the picture. Here's the record of the add process and wireshar capture between the NAS (192.168.10.5) and the CAM (192.168.10.20)
[add onvif camera.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5568964/add.onvif.camera.zip)
- QVE Monitor app for the SS can eventually recognize PTZ and move the camera, but again no video. I have a record for this, but haven't performed traffic capture this time
[add onvif camera.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5568971/add.onvif.camera.zip)
- the workaround is to add the camera twice, one entry for PTZ, other for the RTSP stream

Is there anything else I can help you to resolve the issue in case you're interested? I have an open ticket with Qnap support, but we haven't made any valuable progress there. So if you have something to address to them, I can do that for you.

Thanks!

roleoroleo, 2020-11-19:
Onvif is a strange world.
I will check.it.

roleoroleo, 2020-11-20:
Try this binary:
[onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5572205/onvif_srvd.gz)

Cho0Cho0, 2020-11-20:
Super, it works now.

Let me share the settings if somebody needs that.

Camera still passes test with the question mark, but works fine

[image]

[image]

[image]

These are the yi-hack settings:

System
`HTTPD=yes
TELNETD=yes
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=yes
RTSP_AUDIO_NR_LEVEL=0
ONVIF=yes
ONVIF_WSDD=yes
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
`

Camera
`SWITCH_ON=yes
SAVE_VIDEO_ON_MOTION=no
SENSITIVITY=low
BABY_CRYING_DETECT=no
LED=yes
ROTATE=no
IR=yes`

MQTT is not used.

Camera load is following:
Uptime | 0h 22m 29s
Load Average | 2.23 2.73 2.12
Free/Total Memory | 17664/60824 KB

Sound recording using SS is not really useful as it sounds like the slow-motion and low pitch, despite the picture is quite responsive for my tasks.

P.S. Thanks again and take a beer from me 👍

roleoroleo, 2020-12-04:
Released in 0.3.9.

closed 2020-12-19 by roleoroleo