roleoroleo/yi-hack-MStar #278: Can't reach YI 1080p Home web interface

Rensgiel, 2020-11-21:
I can't reach YI 1080p Home web interface after installing the firmware. 

The specs of my camera: YI 1080P Home - serial: 9FUS*

According to the steps here I formatted a 16GB SD card and put the files home_y203c and sys_y203c on it. After restarting the camera, the restart only took 1 minute. I didn't saw the yellow light flashing. So not sure if the firmware is being flashed succesfully. 
After restarting I was able to access the cam via the YI Home app but wasn't able to reach the camera via http://192.168.2.42:8080. After removing the SD card I can see that there are clips in de record folder on the SD card. Pinging to the IP is succesfull.

I'm stuck now what to do. I have a few questions; is the firmware installed correctly? How can I check that? And what could be the problem that I cannot reach the web interface to change config?

Thanks for helping

roleoroleo, 2020-11-21:
What's your fw version?

Rensgiel, 2020-11-22:
According to the YI Home app it's running 8.2.0.0A_202007211802

roleoroleo, 2020-11-22:
According to supported cameras table, home_y203c and sys_y203c require firmware 4.5.0.
See this link:
https://github.com/roleoroleo/yi-hack-Allwinner

Rensgiel, 2020-11-22:
> According to supported cameras table, home_y203c and sys_y203c require firmware 4.5.0.
> See this link:
> https://github.com/roleoroleo/yi-hack-Allwinner

Thanks, this works!

closed 2020-11-22 by Rensgiel