roleoroleo/yi-hack-MStar #280: Cam blocked

mcgnai, 2020-11-23:
Hi, I have 8 Y3 cams and it has been running for over 5 months with no problems. ( Now with fw 0.3.7) One day I realized that one of these was no longer sending the video and I realized that it was constantly in reset (it turned on, it ticked and then the yellow led started to flash and then it stayed again). I took it. I removed the SD card and reset it using the reset button. I started the procedure to add a new cam on the "YI Home" app from the beginning but it was not successful. I also tried to flash the firmware again (0.3.6) but it does not read the sd because the cam must first be initialized with the YI Home app. I wrote to the supplier who replaced the cam. A month has passed and the same problem has occurred. Can you help me solve?

mcgnai, 2020-11-24:
Thanks for the reply. Can you help me get the original firmware back? where can i download the original fw? what is the procedure? Thanks

roleoroleo, 2020-11-24:
What's your model suffix?

mcgnai, 2020-11-24:
[image: 20201124_110609.jpg]

roleoroleo, 2020-11-25:
I need to know if your cam is y203c, h201c, etc...

janekjelonek, 2020-11-25:
Hi, i believe I've just experienced the same issue. I was on 0.3.6 y203c (Yi Home 1080p / 9FUSY21ZZS). It happened excactly the same way, out of a sudden after a months of service, the cam led turned yellow. I removed SD and performed camera reset.  Then I tried to pair it with YiHome App. The pairing process fails after the cam announces 'Wifi Connected'. The cam is indeed connected to wifi but pairing process fails. 
What is interesting, if I input SD card with yi-hack and perform hardware reset via button (version 0.3.6 or 0.3.7) it boots up to yi-hack and yi-hack works fine. But the state is somewhat like this: 

- cam is on 
- wifi connected 
- yi hack working, 
- unable to pair with yi home app, 
- cam not visible in app, 

in this scenario, after power loss the cam goes back to 'welcome ti yi home camera, waiting to connect...' message as it looses wifi config and credentials. If i want to use it again, I need to perform a reset and start unsuccessful pairing process  with yi home app till it connects to wifi.

Saving wifi settings for good would probably give the cam another life as you can still use it without Yi Home App. 
Yi Customer Service (helpdesk) doesn't know how to solve pairing problems and offers cam replacement. 
Regards, Janek

mcgnai, 2020-11-25:
Thanks for your feedback.  for a moment I thought it was me the problem is
that I couldn't solve it.  your situation is basically the same as mine.  I
hoped to be able to flash the original YI firmware to the cam but it is not
possible because the cam when started with the sd does not read it as it
did when we installed the fw hack and then starts the installation
procedure and with the YI home app again  which is not successful.  The Yi
Customer Service has already replaced me with a cam that had the same
problem but also with this the same thing happened and you have to know
what the problem is.  the strange thing is that I have 7 other cams that
work well at least until today.  if you have new update me.  thanks
Regards
Ste.

roleoroleo, 2020-11-25:
I think your problem could be similar to this one: https://github.com/roleoroleo/yi-hack-MStar/issues/226
When the cam is "blocked" please check the free space.
But if you upload a different hack version (or restore the original fw) the cam MUST start to work again.

mcgnai, 2020-11-25:
ok, the cam model is y25 . I  have already tried to flash the cam with the
version. y25_0.3.6 but when I insert the sd to start the cam, this does not
start the firmware update but begins with the setup procedure to be done
with the YI Home app that does not go well.
Thank's

roleoroleo, 2020-11-25:
Try to change the sd card.
The upgrade procedure is made by the bootloader, before the setup procedure could start.

mcgnai, 2020-11-25:
this is the problem, the cam cannot load any firmware from the sd card
because it must first be initialized from the Yi Home app.  how can you do
to install the original firmware or another version of hack?  if you want I
can send you a movie

mcgnai, 2020-11-25:
I have already tried to change the sd but nothing, the bootloader does not
start but the setup procedure begins.

mcgnai, 2020-11-25:
Can you send me the file of original fw because is not present on the YI official site?Thank'sInviato dal mio Galaxy
-------- Messaggio originale --------Da: Stefano Mugnai <mcgnai@gmail.com> Data: 25/11/20  21:12  (GMT+01:00) A: roleoroleo/yi-hack-MStar <reply@reply.github.com> Oggetto: Re: [roleoroleo/yi-hack-MStar] Cam blocked (#280) I have already tried to change the sd but nothing, the bootloader does not start but the setup procedure begins.

ethempekin, 2020-11-25:
My Yi Dome 1080 is hacked and running fm 0.3.5 for model h201c. I am still accessing it on homebridge but there seems to be a problem with yi home app. Cannot add it there. It used to be there but was not responding so now removed it. Adding through the app time outs after entering wifi connection. How can I solve this issue?

mcgnai, 2020-11-26:
> I need to know if your cam is y203c, h201c, etc...

ok, the cam model is y25 . I  have already tried to flash the cam with the
version. y25_0.3.6 but when I insert the sd to start the cam, this does not
start the firmware update but begins with the setup procedure to be done
with the YI Home app that does not go well.
Thank's

roleoroleo, 2020-11-26:
@mcgnai 
Here it is:
[y25.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5603480/y25.tar.gz)
But I think this will not solve your problem.

denven, 2020-11-30:
I have the same issue since last night, the problem came after I fixed a [google drive uploading issue](https://github.com/roleoroleo/yi-hack-MStar.gdrive/issues/5#issuecomment-735353515):

- My video files failed to upload to google drive again(I used another [gdrive hack](https://github.com/roleoroleo/yi-hack-MStar.gdrive)), at first I thought I didn't solve the issue, however, when I use my own computer(Linux) and use the same script to upload video files, it works.
- So I tried to reconfigure google API settings, but I found [curl not working](https://github.com/roleoroleo/yi-hack-MStar.gdrive/issues/6), but I can ping google DNS IP from the SSH console, but cannot resolve a name service. I thought it's my camera network problem.
- Then I reset my camera and hope I could get a brand new environment, however, I can never connect my phone App to the camera again, and the App cannot pair with the camera.
- Currently, the camera cannot be connected and managed from the Yi App. I checked the video files in record directory, it stopped to record since the time when network broke. 
- And I cannot re-install this hack from the SD card.

- Yi Dome 1080p
- BFUS**
- Firmware: 4.6.0.0A 201908271549 ---> this hack

I believe why the Yi App cannot pair with the camera is due to the camer is:
- kind of offline because of camera network issue (cannot resolve name service)
- or blocked 

The way I think it may fix this issue: downgrade the firmware to the official one.

However, I cannot find the latest official firmware version, if someone can find the latest firmware, please leave a comment here.
 and is there a way to debug the camera from USB, cause I cannot find the IP address of the camera(the router is not mine).

denven, 2020-11-30:
> I think your problem could be similar to this one: #226
> When the cam is "blocked" please check the free space.
> But if you upload a different hack version (or restore the original fw) the cam MUST start to work again.

😓 If I have read this earlier, my camera may be working right now.

roleoroleo, 2020-11-30:
I don't know if I can fix this problem.
I already removed all unnecessary applets from busybox and I can't increase the free space on /.

vkoskela, 2020-12-02:
I had also the app pairing problem and downgrade to 0.3.7 worked for me 👍

roleoroleo, 2020-12-04:
I moved all variable files to /home partition.
I think it should fix the problem.
Try 0.3.9

denven, 2020-12-04:
> I moved all variable files to /home partition.
> I think it should fix the problem.
> Try 0.3.9
Thanks, I just upgraded the new version.  The /dev/root free space is 16KB, is this right?
Before my upgrading, the free space is only 4K.
```
/home # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.9M      1.9M     16.0K  99% /
tmpfs                    29.7M      1.8M     27.9M   6% /dev
/dev/block/mtdblock3     12.2M     10.8M      1.4M  89% /home
tmpfs                    32.0M    100.0K     31.9M   0% /tmp
/dev/block/mmcblk0p1     29.7G     47.9M     29.7G   0% /tmp/sd
```

roleoroleo, 2020-12-04:
Free space depends on the specific model (suffix).
In my case (y203c) is about 30KB.
But I moved to /home all variable files:
- /etc/TZ
- /etc/hostname
- /etc/passwd

And I removed back.bin from the original fw.

roleoroleo, 2020-12-19:
Could you send me a feedback about this issue?
Is it still present in version 0.3.9?

roleoroleo, 2020-12-27:
News about this issue?

closed 2021-02-11 by roleoroleo

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-03-18:
Just wanted to say I have just encountered a similar issue and downgrading from 0.3.9 to 0.3.7 fixed it for now.
(Yi 1080p Dome BFUS h201c).  Was running 0.3.9 since December 2020 and only encountered issue since yesterday.