roleoroleo/yi-hack-MStar #281: Kami Yi external cam compatibility

nicmela, 2020-11-24:
Hey, 
I recently bought a Kami Wire-free outdoor Camera ( https://www.yitechnology.com/kami-wirefree ). Are this firmwares compatible with this cam?

roleoroleo, 2020-11-24:
What's firmware version?

nicmela, 2020-11-24:
Not sure if I'm taking it from the right place, searching in Kami Home app I can see in the cam options: 
Actual version: 5.1.0.0A_202004022041

roleoroleo, 2020-11-25:
At the moment is not supported.
To check if it's possible to support it in the future, I need a dump of the memory. See the wiki.

liraneli27, 2020-11-25:
Hi,
same question here , only that i bought the kami wire free home security (no sd slot but an mini usb port on the camera)
(https://kamihome.com/wirefree-wk101)

even if the firmwares are compatible - there's a way to update without an SD slot?
when i plug it to my computer nothing happens :|

nicmela, 2020-11-25:
Okay so... not a disassemble expert here... I managed to open the top lid (hard task not knowing what to do and what can break) and tried to find the 3 uart pads.

[image]
[image]

Can you help me identify them? Also I will need to buy an adapter, will this work? https://www.amazon.it/Serial-PL2303HX-Adattatore-RS232-download/dp/B07QX66Y95 
[image]

Thank you!

roleoroleo, 2020-11-26:
Sorry but from these pictures is not possible to find uart pads.
And it's not possible to check the cpu (Hisilicon, Mstar, Allwineer, etc...).

nicmela, 2020-11-26:
Aight... it was tricky... but i managed to deasseble it more to checkout the hidden side. Does this helps?
[image]

roleoroleo, 2020-11-26:
2 news:
1.  Above the chip there are the pads where you can connect the uart adapter: RXD and TXD. GND you can choose the one you want.
2. This cam is based on hisilicon chip, this hack is not compatible.

nicmela, 2020-11-26:
AIght.. thank you for your time!
I've checked TheCrypt0 project, but seems pretty much dead. Sad :(

thoemmes86, 2020-12-01:
I would also support and donate a hack for this camera.
Since black week I have two of those cameras which are really nice. 
But like normal for yi, no support for external IOT Systems.

nicmela, 2020-12-01:
I definitely would also support and donate for this.

roleoroleo, 2020-12-01:
I can't support it.
The hisilicon platform is completely different.

closed 2020-12-19 by roleoroleo