roleoroleo/yi-hack-MStar #282: 48US support

alienatedsec, 2020-11-25:
Hi All

I understand there is another project covering models with hi3518e chipset, but I was wondering if there would be a chance to integrate ONVIF and steps that I would need to follow. Unfortunately, the other projects seem to be dead since early last year; hence why my question over here as more activity and commitment has been recognised. 

Any help or direction would be appreciated.

roleoroleo, 2020-11-25:
I can't build a new hack for this model but I could help you to do it, if you want.

alienatedsec, 2020-11-25:
> I can't build a new hack for this model but I could help you to do it, if you want.

If that is an option, I am keen to even contribute in helping the community. Let me know how you want to approach it.

roleoroleo, 2020-11-26:
First step: clone the repo and rebuild a whole hack.

alienatedsec, 2020-11-26:
@roleoroleo which stock firmware you want me to use? I would only have the one for hi3518e at the moment.

roleoroleo, 2020-11-26:
Which hack are you using at the moment?
yi-hack-v4 from TheCrypt0?

Start cloning TheCrypt0 repo and use hi3518e firmware.
I know that v4 hack is compatible with different models: y18, y20, v201, h20, y19 and h30 and probably you don't have stock firmwares for all models.
Start with what you have.

alienatedsec, 2020-11-26:
@roleoroleo for some reason I am getting this error related to rootfs_y20.
[image]

Any ideas?

roleoroleo, 2020-11-26:
Probably rootfs_y20 is not the correct format.
Try to mount it manually with this script:
```
#!/bin/bash

## Script to mount jffs2 filesystem using mtd kernel modules.
## EMAC, Inc. 2009

if [[ $# -lt 2 ]]
then
    echo "Usage: $0 FSNAME.JFFS2 MOUNTPOINT [ERASEBLOCK_SIZE]"
    exit 1
fi

if [ "$(whoami)" != "root" ]
then
    echo "$0 must be run as root!"
    exit 1
fi

if [[ ! -e $1 ]]
then
    echo "$1 does not exist"
    exit 1
fi

if [[ ! -d $2 ]]
then
    echo "$2 is not a valid mount point"
    exit 1
fi

if [[ "$3" == "" ]]
then
<------>esize="128"
else
<------>esize="$3"
fi

# cleanup if necessary
umount /dev/mtdblock0 &>/dev/null
modprobe -r mtdram &>/dev/null
modprobe -r mtdblock &>/dev/null

modprobe mtdram total_size=32768 erase_size=$esize || exit 1
modprobe mtdblock || exit 1
dd if="$1" of=/dev/mtdblock0 || exit 1
mount -t jffs2 /dev/mtdblock0 $2 || exit 1

echo "Successfully mounted $1 on $2"
exit 0
```

Command:
`mount.jffs2 image.jffs2 mount_dir 64`

alienatedsec, 2020-11-26:
[image]

roleoroleo, 2020-11-26:
I don't know.
Try to restart from the beginning.

alienatedsec, 2020-11-26:
Does it matter if '/tmp/ssc' is empty?

roleoroleo, 2020-11-26:
No.
Looking at the script
https://github.com/TheCrypt0/yi-hack-v4/blob/master/scripts/init_sysroot.sh
it looks like there is a 64 byte header before the jffs2 image.

alienatedsec, 2020-11-26:
I loaded those stock files through 'yi-hack-v4' web interface, and the camera is now running the stock firmware, so both files have to be correct.

I also mounted home_y20 the same way, and I could see some files.
[image]

Starting again gives the same results. Any further ideas?

roleoroleo, 2020-11-26:
Try to run manually all commands inside the script.
Step by step.
We will check where it stops working.

alienatedsec, 2020-12-01:
Just to update, I still have it on my to-do list, but there was no further progress since.

However, would I need the SDK for hi3518e installed properly in order to successfully mount or even process the init_sysroot.sh?

roleoroleo, 2020-12-01:
Yes, check this wiki:
https://github.com/TheCrypt0/yi-hack-v4/wiki/Project-Setup-Hints

alienatedsec, 2020-12-01:
[image]

Despite being successful with the SDK, it has the same error when executing the script for rootfs file

roleoroleo, 2020-12-01:
Decompress manually the file using the script I posted above and copy the file in the sysroot folder.
If the script finds the file in the sysroot folder, it doesn't try to decompress the image.

alienatedsec, 2020-12-01:
> Does it matter if '/tmp/ssc' is empty?

It was empty when I did it (mounted rootfs) the last time.

> No.
> Looking at the script
> https://github.com/TheCrypt0/yi-hack-v4/blob/master/scripts/init_sysroot.sh
> it looks like there is a 64 byte header before the jffs2 image.

The content of the mounted folder is empty even if I mount yi-hack-v4 'rootfs_y20' file, but it is ok for 'home_y20'.

Running everything in verbose mode seems to be working great until it mounts the 'rootfs_y20' and tries to copy something which does not exist; hence, the error.

roleoroleo, 2020-12-02:
Mount with this:
https://github.com/roleoroleo/yi-hack-MStar/issues/282#issuecomment-734426847
and copy the files manually.

alienatedsec, 2020-12-02:
@roleoroleo Which version of mtd-utils you have?

[image]

roleoroleo, 2020-12-02:
1:2.0.0-1 on Debian 9.

> Does it matter if '/tmp/ssc' is empty?

Now, I understand the question.

Probably the image is corrupted. Are you sure that the image works when you try to restore the stock firmware.
Maybe home_y20 is processed and applied but sys_y20 isn't.
So, you think you have restored the whole stock firmware but you restored only the home partition.

alienatedsec, 2020-12-02:
> Probably the image is corrupted. Are you sure that the image works when you try to restore the stock firmware.
> Maybe home_y20 is processed and applied but sys_y20 isn't.
> So, you think you have restored the whole stock firmware but you restored only the home partition.

I have tried to mount plenty of 'rootfs_x20' files found in the universe (google) and all of them mounted with no content. home_x20 are fine **but rootfs_x20 are empty.**

Sounds like a lame issue, but I tried everything except Debian 9 ;)

roleoroleo, 2020-12-02:
Did you try to change the eraseblock size?

alienatedsec, 2020-12-03:
> Did you try to change the eraseblock size?

Multiple times.

roleoroleo, 2020-12-04:
Share with me a link of this file.
I try to decompress it.

alienatedsec, 2020-12-04:
Thank you @roleoroleo for your help - sounds like I am doing something wrong anyway - [LINK](https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/32280606850)

alienatedsec, 2020-12-04:
The latest Ubuntu version and its mtd-utils have the same problem.

[image]

roleoroleo, 2020-12-04:
With my debian 9 works without problems.
Here it is:
[rootfs.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5642128/rootfs.tar.gz)

alienatedsec, 2020-12-04:
> With my debian 9 works without problems.

Thanks. I will be doing the same on Debian. I started it already wrong as installed Debian 10 instead of 9 as a docker.
I will let you know how it went

roleoroleo, 2020-12-04:
Meanwhile, if you extract rootfs.tar.gz in sysroot, the init script will not try to decompress home_y20.

alienatedsec, 2020-12-04:
[image]

Debian 9, mtd-utils v2.0.0 - anything above that fails

The next steps please @roleoroleo

roleoroleo, 2020-12-04:
Try to build the hack.
```
./scripts/compile.sh
```

alienatedsec, 2020-12-05:
> Try to build the hack.
> 
> ```
> ./scripts/compile.sh
> ```

I forgot about the SDK, so....

Now, I have a problem which didn't exist in Ubuntu - This [guide](https://www.programmersought.com/article/933051736/) - point 4.2 

[image]

Although the cross-compilation toolchain is installed, it still gives me the same error on Debian (it worked fine on Ubuntu). 
@roleoroleo **What is different in Debian? Maybe the below - Step 10?**

> Step 10, vi ~/.bashrc, add export PATH=/opt/hisi-linux/x86-arm/arm-hisiv300-linux/target/bin:$PATH at the end of the file.

Here is the command confirming the successful installation
`arm-hisiv300-linux-gcc -v`

[image]

Here is the error after running the code
`make OSDRV_CROSS=arm-hisiv300-linux CHIP=hi3518ev200 all`

[image]

roleoroleo, 2020-12-05:
The line 83 of the compile script set the path:
`export PATH="/opt/hisi-linux/x86-arm/arm-hisiv300-linux/target/bin:$PATH"`
If you copy the toolchain in this path it should work.

Without editing bashrc or setting environment variables.

alienatedsec, 2020-12-05:
> The line 83 of the compile script sets the path:
> `export PATH="/opt/hisi-linux/x86-arm/arm-hisiv300-linux/target/bin:$PATH"`
> If you copy the toolchain in this path it should work.
I am not sure what to copy where, as 'arm-hisiv-linux*' files are already in the above location.
> 
> Without editing bashrc or setting environment variables.
Those are also correct, but for some reason, it doesn't recognise the command when executed by the compile script

roleoroleo, 2020-12-05:
There is a problem.
The prefix of the toolchain is different.
The archive contains the toolchain with prefix arm-hisiv300-linux-uclibcgnueabi
The hack requires the toolchain with prefix arm-hisiv300-linux-
Probably you need to change the hack scripts.

roleoroleo, 2020-12-05:
For example
Change the path in compile.sh
`export PATH="/opt/hisi-linux/x86-arm/arm-hisiv300-linux/bin:$PATH"`
Change the cross compiler in busybox/.config
`CONFIG_CROSS_COMPILER_PREFIX="arm-hisiv300-linux-uclibcgnueabi-"`

roleoroleo, 2020-12-05:
Did you forked the repo?
Fork the repo and add my user to the contributor list. I will help you.

alienatedsec, 2020-12-05:
> Did you forked the repo?
> Fork the repo and add my user to the contributor list. I will help you.

@roleoroleo Its Done

roleoroleo, 2020-12-05:
Ok, now you should be able to build it.
Copy the toolchain in /opt

alienatedsec, 2020-12-11:
Hi @roleoroleo again

Apologies, I had a little break from .... everything to be fair.

I don't think it is correct to amend the 'compile.sh' script from the repo as the process fails at the SDK compilation stage.

The error appears after running the command from [this guide](https://www.programmersought.com/article/933051736/)
`make OSDRV_CROSS=arm-hisiv300-linux CHIP=hi3518ev200 all`

I can see what you did with my forked repo, but I wasn't even at the stage to compile the firmware. Let me check a few things, and I will come back with more information.

roleoroleo, 2020-12-11:
> I don't think it is correct to amend the 'compile.sh' script from the repo as the process fails at the SDK compilation stage.

You don't have to compile sdk. It's alreay compiled.
If you decompress the archive linked from TheCrypt0 (https://drive.google.com/open?id=1WvjkkItAOZRxBm6mfvs8KEmQWffcuCsS) you can find all binaries.
Simply copy the folder in /opt
The path is:
Hi3518E_SDK_V1.0.4.0/package/osdrv.tgz/osdrv/opensource/toolchain/arm-hisiv300-linux/arm-hisiv300-linux.tar.bz2/arm-hisiv300-linux

> I can see what you did with my forked repo, but I wasn't even at the stage to compile the firmware. Let me check a few things, and I will come back with more information.

I tested the repo and it works correctly.

alienatedsec, 2020-12-11:
@roleoroleo you were right - my mistake

Now, is there any reason why it doesn't go any further
[image]

roleoroleo, 2020-12-11:
Initialize submodules.

alienatedsec, 2020-12-11:
@roleoroleo its done. You are a patient guy, I can tell :) Thanks for your help

The next step is to load it. I need to find my 48US model.

[image]

alienatedsec, 2020-12-11:
It keeps rebooting for some reason. I will recompile it - Is there anything I have to be aware of? 

Please see the below 'compile.sh' output.
[compilelog.txt](https://github.com/roleoroleo/yi-hack-MStar/files/5681313/compilelog.txt)

roleoroleo, 2020-12-11:
I added onvif_srvd and wsdd but I don't know if there is enough space in the the flash.
So try to build first removing src/onvif_srvd and src/wsdd folders.

roleoroleo, 2020-12-11:
I don't know very well this platform but, if it is the same as the others, it's not possible to have a log.
You should create a special image with your debug system.

roleoroleo, 2020-12-19:
News about your project?

alienatedsec, 2020-12-31:
Hi @roleoroleo

Thanks for your message.

I have tried a few things, but it was behaving the same. I still have to delete those folders and load the firmware. It's a festive period, so I stopped working on it for now. I am looking forward to restarting my further testing in the week commencing 4th January. 

TheCrypt0 announced some updates today and the ONVIF integration in the next couple of weeks - that will be interesting, considering there was no update in the last 19 months

idollark, 2021-01-13:
Hello,

@roleoroleo ... Thanks a lot for your work. Indeed !
@alienatedsec ... Thanks a lot also. It seems that you have been trying to solve my issue.

I have a YI dome 45US

> /home/yi-hack-v4/www/cgi-bin # cat /proc/cpuinfo 
> Processor	: ARM926EJ-S rev 5 (v5l)
> BogoMIPS	: 269.10
> Features	: swp half fastmult edsp java 
> CPU implementer	: 0x41
> CPU architecture: 5TEJ
> CPU variant	: 0x0
> CPU part	: 0x926
> CPU revision	: 5
> 
> Hardware	: hi3518ev200
>
> Revision	: 0000
> Serial		: 0000000000000000

I have installed yi-hack-v4 but it the hack does not offer "/home/yi-hack/bin//ipc_cmd" which I need to move the camera.

The "yi-hack for MStar platform", which I am running in a different camera, is my perfect solution (thanks  @roleoroleo  !!!)

I could not understand the status of the work @alienatedsec ?
Should I expect a extension of is hack soon ?

Many thanks for your work and the answer

alienatedsec, 2021-01-16:
> I could not understand the status of the work @alienatedsec ?

Hi  @idollark 

I aimed to integrate functionalities from yi-hack-MStar into yi-hack-v4 to support the legacy chipset. Especially that yi-hack-v4 is poorly maintained. I must admit that @roleoroleo is doing a great job with yi-hack-MStar, which I wanted to migrate, but it is almost impossible to achieve with my limited time. Let's hope I could get back to review it at some point this weekend. 

However, no matter how long it will take me, it will be quicker than the next yi-hack-v4 release. (No matter what I think, I would like to be very wrong about this statement)

idollark, 2021-01-16:
> Hi @idollark
>
> I aimed to integrate functionalities from yi-hack-MStar into yi-hack-v4 to support the legacy chipset. Especially that yi-hack-v4 is poorly maintained. I must admit that @roleoroleo is [...]

Hello and thank you again,

It sounds like a good thing to do. Actually, I will be very happy to help.
My time is also limited but I am willing to find some.
Is there a way to get in contact without posting public ? I guess that there is no need to discuss details here. Of course, the result shall be public,

Cheers

alienatedsec, 2021-01-23:
I am back in the game @roleoroleo - at least for today :)

However, I noticed that something is not compiling right. Is the content of the below folder correct after running 'compile firmware' step?

`./scripts/compile.sh`

[image]

alienatedsec, 2021-02-02:
Hi @roleoroleo 

Can you please confirm in your free time if the folder content is correct after 'compile'?

roleoroleo, 2021-02-02:
It's not correct.
Probably the compilation process exits with errors.

alienatedsec, 2021-02-13:
> It's not correct.
> Probably the compilation process exits with errors.

@roleoroleo 
The error is not obvious, but nothing is happening after the libfuse compilation step as per the below. It is the same on Debian 9 and Ubuntu 18.04 LTS. Any ideas about how to handle it further?
[image]

roleoroleo, 2021-02-13:
Probably libfuse is compiled with the "standard" gcc instead off cross gcc.
Check if the toolchain is ok and is installed in the correct path: /opt/arm-hisiv300-linux/bin

Another strange detail is the name: libfuse3.so.3.4.2.p
Why the last "p"?
Try to clone again the repo and rebuild it.

alienatedsec, 2021-02-13:
The path is correct as I tried to compile it for both TheCrypt0 'yi-hack-v4' and my fork with your toolchain location amendment.

[image]

 

> Another strange detail is the name: libfuse3.so.3.4.2.p
> Why the last "p"?

Maybe the reason for the folder name?
[image]

There is a config file named `Hi3518eV200.conf` in the `src/libfuse` folder, which seems to be also correct.

[image]

alienatedsec, 2021-02-13:
There are also files in the `target/bin` folder:

[image]

[45 more replies]

closed 2021-02-23 by alienatedsec