roleoroleo/yi-hack-MStar #291: [Question] Create time lapse

joekerna, 2020-12-07:
I have a quick question: Is it possible to create a time lapse using the Yi camera?

I would like to take a picture every ~30 seconds for about 8h using only the camera and a USB Powerbank?!

roleoroleo, 2020-12-07:
I know ffmpeg is able to do time lapse.
But the binary is too big to fit into this hack.

joekerna, 2020-12-08:
Do you think I could put it on the SD card?

roleoroleo, 2020-12-08:
Yes, copy it in /tmp/sd/yi-hack/bin
[ffmpeg.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5659568/ffmpeg.gz)

Rushmed, 2020-12-25:
Pls. provide tutorial of how to create timelapse.
Thanks in further an am merry christmas!

roleoroleo, 2020-12-27:
I didn't try it but this page sould solve your question.
https://superuser.com/questions/1499968/creating-timelapse-from-still-images-jpg-to-mp4-using-ffmpeg

joekerna, 2020-12-27:
Thanks, I've seen that. This page describes how to generate a movie from JPEGs using `ffmpeg`.

I am still looking for a way to tell the camera to automatically take a photo every x seconds after booting. Creating a film from those photos I can still do later.

roleoroleo, 2020-12-29:
Create your own directory on the SD card and schedule this command with crontab:

/home/yi-hack/bin/imggrabber -r high -w > /tmp/MY_DIR/ss_`date +"%H_%M"`.jpg

joekerna, 2021-01-03:
Do I have to activate cron somehow? I've added

> 26 10 * * * /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`date +"%H_%M_%S"`.jpg

to my crontab but after 10:26 no photo was taken?! I don't know if the cronjob is being executed.

joekerna, 2021-01-03:
I manually started `crond` and after that the cronjob was executed. Now I only have to autostart `crond`

roleoroleo, 2021-01-03:
crond should start automatically starting from 0.3.9.
Simply add your command to the crontab section in configuration page and reboot the cam.

joekerna, 2021-01-03:
I didn't know about that feature in the web-config. Thanks!

I've entered 

``* * * * * /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`/bin/date +'%H_%M_%S'`.jpg``

but it is not working quite yet. There is an issue with the timestamp. If I assign the same name every time it works.

roleoroleo, 2021-01-03:
You are right, I found a bug with quote char.
Apply the following changes:
https://github.com/roleoroleo/yi-hack-MStar/commit/fb1c8c1f18b021aa6429cf4df75681d8223ec272

joekerna, 2021-01-03:
Awesome! That's working. Now I will try to get a granularity smaller than 1 minute by doing something like [this](https://linuxconfig.org/how-to-execute-less-than-1-minute-intervals-jobs-using-cron-time-based-scheduler)

joekerna, 2021-01-03:
I'm running the following script every minute using cron:

```
#!/bin/sh

          date >> /tmp/sd/cron_test
sleep 10; date >> /tmp/sd/cron_test
sleep 20; date >> /tmp/sd/cron_test
sleep 30; date >> /tmp/sd/cron_test
sleep 40; date >> /tmp/sd/cron_test
sleep 50; date >> /tmp/sd/cron_test
```

However a `cat /tmp/sd/cron_test` shows:

```
cat /tmp/sd/cron_test 
Sun Jan  3 21:08:10 CET 2021
Sun Jan  3 21:08:30 CET 2021
Sun Jan  3 21:09:00 CET 2021
Sun Jan  3 21:09:40 CET 2021
Sun Jan  3 21:10:30 CET 2021
Sun Jan  3 21:11:10 CET 2021
Sun Jan  3 21:11:30 CET 2021
Sun Jan  3 21:12:00 CET 2021
Sun Jan  3 21:12:40 CET 2021
Sun Jan  3 21:13:30 CET 2021
Sun Jan  3 21:14:00 CET 2021
Sun Jan  3 21:14:10 CET 2021
Sun Jan  3 21:14:30 CET 2021
Sun Jan  3 21:15:00 CET 2021
Sun Jan  3 21:15:00 CET 2021
Sun Jan  3 21:15:10 CET 2021
Sun Jan  3 21:15:30 CET 2021
Sun Jan  3 21:15:40 CET 2021
Sun Jan  3 21:16:00 CET 2021
Sun Jan  3 21:16:30 CET 2021
```

It is skipping quite a few exections?!

joekerna, 2021-01-04:
```
#!/bin/sh

(          date >> /tmp/sd/cron_test) &
(sleep 10; date >> /tmp/sd/cron_test) &
(sleep 20; date >> /tmp/sd/cron_test) &
(sleep 30; date >> /tmp/sd/cron_test) &
(sleep 40; date >> /tmp/sd/cron_test) &
(sleep 50; date >> /tmp/sd/cron_test) &
```

This is working like it's supposed to:
```
cat /tmp/sd/cron_test 
Mon Jan  4 14:25:00 CET 2021
Mon Jan  4 14:25:10 CET 2021
Mon Jan  4 14:25:20 CET 2021
Mon Jan  4 14:25:30 CET 2021
Mon Jan  4 14:25:40 CET 2021
Mon Jan  4 14:25:50 CET 2021
Mon Jan  4 14:26:00 CET 2021
Mon Jan  4 14:26:10 CET 2021
Mon Jan  4 14:26:20 CET 2021
Mon Jan  4 14:26:30 CET 2021
Mon Jan  4 14:26:40 CET 2021
Mon Jan  4 14:26:50 CET 2021
Mon Jan  4 14:27:00 CET 2021
Mon Jan  4 14:27:10 CET 2021
Mon Jan  4 14:27:20 CET 2021
Mon Jan  4 14:27:30 CET 2021
Mon Jan  4 14:27:40 CET 2021
Mon Jan  4 14:27:50 CET 2021
Mon Jan  4 14:28:00 CET 2021
Mon Jan  4 14:28:10 CET 2021
Mon Jan  4 14:28:20 CET 2021
Mon Jan  4 14:28:30 CET 2021
Mon Jan  4 14:28:40 CET 2021
Mon Jan  4 14:28:50 CET 2021
Mon Jan  4 14:29:00 CET 2021
Mon Jan  4 14:29:10 CET 2021
Mon Jan  4 14:29:20 CET 2021
Mon Jan  4 14:29:30 CET 2021
Mon Jan  4 14:29:40 CET 2021
Mon Jan  4 14:29:50 CET 2021
Mon Jan  4 14:30:00 CET 2021
Mon Jan  4 14:30:10 CET 2021
Mon Jan  4 14:30:20 CET 2021
Mon Jan  4 14:30:30 CET 2021
Mon Jan  4 14:30:40 CET 2021
Mon Jan  4 14:30:50 CET 2021
Mon Jan  4 14:31:00 CET 2021
Mon Jan  4 14:31:10 CET 2021
Mon Jan  4 14:31:20 CET 2021

```

joekerna, 2021-01-04:
It is working:

`* * * * * /home/yi-hack/timelapse.sh`
```

#!/bin/sh

(          /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`/bin/date +'%H_%M_%S'`.jpg) &
(sleep 10; /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`/bin/date +'%H_%M_%S'`.jpg) &
(sleep 20; /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`/bin/date +'%H_%M_%S'`.jpg) &
(sleep 30; /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`/bin/date +'%H_%M_%S'`.jpg) &
(sleep 40; /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`/bin/date +'%H_%M_%S'`.jpg) &
(sleep 50; /home/yi-hack/bin/imggrabber -r high -w > /tmp/sd/ss_`/bin/date +'%H_%M_%S'`.jpg) &
```
I will try a longrun tomorrow and then create an actual film from the JPEGs using `ffmpeg`

joekerna, 2021-01-19:
The whole timelapse process is working fine. I wanted to use it to make a timelapse of a building construction.

However not a single photo was taken at the construction side because there is no WIFI. All my tests were run at home where the is wifi

Why does the camera require wifi to store photos onto the SD card?

roleoroleo, 2021-01-19:
I have never tested this scenario.
Probably the yi processes don't start if a connection is not available.
I will check it.

joekerna, 2021-03-26:
Have you been able to check it? We could take really cool timelapse videos with this if the wifi connection wasn't required

roleoroleo, 2021-03-26:
I didn't test it because I need to reconnect serial cable.
And the cam is in use in this moment.

roleoroleo, 2021-04-02:
Do you want to try this solution?
https://github.com/roleoroleo/yi-hack-Allwinner/issues/234

Try to add a record in /etc/hosts:
echo "127.0.0.1    api.eu.xiaoyi.com" > /etc/hosts

closed 2024-05-10 by github-actions[bot]