roleoroleo/yi-hack-MStar #294: Yi Camera BFUS Firmware version 9.0.09.02_202007141512

r4zi3lrrr, 2020-12-18:
Hi All,

Newbie here so please be gentle :)

I've purchased a Yi Camera BFUS Firmware version 9.0.09.02_202007141512 from Amazon but really want to get this yi hack on.

I've tried copying over the relevant files to SD card (formatted to FAT 32 with 16gb)  but it doesn't seem to do anything - still can't access camera via web-browser.

Yi app works fine.

Please see attached for my SD card layout.

[image]

MD-V, 2020-12-18:
@r4zi3lrrr,

you will have to extract y20ga_0.2.2.tar aswell. 
There have to be two files on the SD card: one starting with sys_* and one with home_*

Also you should consider to use a more recent version (current version is 0.3.9)

roleoroleo, 2020-12-18:
Based on the fw version, your cam is not supported.
Try another hack.

r4zi3lrrr, 2020-12-18:
> Based on the fw version, your cam is not supported.
> Try another hack.

Is there any hack in particular that you can recommend please?

r4zi3lrrr, 2020-12-18:
> @r4zi3lrrr,
> 
> you will have to extract y20ga_0.2.2.tar aswell.
> There have to be two files on the SD card: one starting with sys_* and one with home_*
> 
> Also you should consider to use a more recent version (current version is 0.3.9)

Thank you will give it a try.

r4zi3lrrr, 2020-12-18:
If I build my own firmware using SDK using: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Build-your-own-firmware  would I be able to hack it myself?

roleoroleo, 2020-12-18:
I don't know, it depends on the model.
But are you sure about the fw version number?

r4zi3lrrr, 2020-12-18:
Ahh ok.

Camera is a Yi 1080p Home Camera
I got the firmware off the Yi app  - hopefully that's the right place to get it from.

roleoroleo, 2020-12-18:
Are you sure it isn't 9.0.19.02_202007141512.

r4zi3lrrr, 2020-12-18:
Apologies amended title now 
Firmware is version 9.0.19.02_202007141512

r4zi3lrrr, 2020-12-18:
Can we work with this Firmware version 9.0.19.02_202007141512 to action any Yi hack?

The Model of camera is: Yi 1080p Home Camera BFUS?

roleoroleo, 2020-12-18:
yi-hack-allwinner-v2

r4zi3lrrr, 2020-12-18:
Thanks mate will try this tomorrow.

r4zi3lrrr, 2020-12-19:
Hi Guys,

Extracted to SD as per below screenshot, put the SD card into the Yi camera - left it for several minutes but not getting anything else, just a amber light.

Any ideas, please?

[image]

roleoroleo, 2020-12-19:
Plwase, move your issue to allwinner-v2 repo.

r4zi3lrrr, 2020-12-19:
Now worries will do thanks.

closed 2020-12-19 by r4zi3lrrr

r4zi3lrrr, 2020-12-22:
> Based on the fw version, your cam is not supported.
> Try another hack.

> @r4zi3lrrr,
> 
> you will have to extract y20ga_0.2.2.tar aswell.
> There have to be two files on the SD card: one starting with sys_* and one with home_*
> 
> Also you should consider to use a more recent version (current version is 0.3.9)

Does this apply to the yi-hack-Allwinner-v2 too?