roleoroleo/yi-hack-MStar #301: Question on firmware upgrade

ivityc, 2021-01-06:
Hi,
I currently have a Yi 1080p Home BFUS
Base Version | 4.5.0.0C_201910080934
serial BFUSY21IFYNRM1191xxx
Model Suffix | y203c
current fw_version is 0.3.4
running without SD card inserted.

I have the option in the maintenance page to perform a firmware upgrade and I see there is a new version 0.3.9
https://github.com/roleoroleo/yi-hack-MStar/releases/tag/0.3.9 (y203c_0.3.9.tgz)

can I download such file and then simply upload it via the maintenance page  (without SD card inserted)
or shall I put an empty SD card and start the upgrade process via the maintenance page
or shall I proceed as the initial hack putting the fw directly on the SD card and rebooting the camera?

Sorry to bother but did not found such info on the wiki.
Thanks and regards
Iv.

roleoroleo, 2021-01-07:
You need to insert a FAT32 formatted SD card (empty or not it doesn't matter) and start the upgrade process via the maintenance page.

ivityc, 2021-01-07:
Ok thanks. Once the upgrade is completed , the SD card shall remain inserted forever?  Or can it be then removed? 
( I have also another Yi model yi-hack-Allwinner-v2 and there , probably due to its beta status, I know that SD card shall be always in)

roleoroleo, 2021-01-07:
In this model (MStar) you can remove the sd card.
AllWinner-v2 requires the sd because the file system on flash is different.

ivityc, 2021-01-10:
Ok thanks for the support. I managed to update the firmware and indeed it works afterword even without the sd card.
I guess it is normal that if I enable the video save upon motion detection, the SD card need to be inserted, even if I also upload the footage on external FTP server (with option to delete the events from the cam afterward)

roleoroleo, 2021-01-11:
Yes, you need an SD card to save locally the files before sending to the ftp server.

closed 2021-02-11 by roleoroleo