roleoroleo/yi-hack-MStar #303: Disabling video upon motion not working

joekerna, 2021-01-09:
I've disabled the setting _Save video when a motion is detected_ but the camera still created a directory named `record` and saves many many video files in it.

Is that a misunderstanding or a bug?

roleoroleo, 2021-01-16:
It's a misunderstanding.
If enabled, video will be saved ONLY when a motion is detected.
If disabled, video will be ALWAYS saved.

arthur0412, 2021-01-17:
I would like to implement the following scenario:
- video is enabled
- motion detection is disabled (because i will use an externl motion detection device)
- save video "always" is disabled
- trigger video save via HTTP request record.sh?time=TIME from my home automation

So what i need is an option to 
- enable "cam" 
- disable "motion" 
- disable "always save video"

Is it possible to implement this?

My current workarund is:
default settings:
- cam off
- video detection on
- detection senitivity "low"

Now i have an script which one did following:
- switch cam on
- trigger save video via HTTP request record.sh?time=TIME (for excample 20sec.)
- wait 20sec. + 1-2sec.
- switch cam off
so finaly i have an 20sec. video.

roleoroleo, 2021-01-17:
The easiest way is to use a script as you are already doing.
The script can be local (a shell script in the cam) or remote using webservices.
Pay attention: "enable/disable cam" feature is not available in the cam (only in the yi app).
To configure the cam using ws see the following wiki: https://github.com/roleoroleo/yi-hack-MStar/wiki/Web-services-description#camera-settings

arthur0412, 2021-01-17:
Yes correct, but what I mean with "cam disable/enable or off/on" is the option "Switch on/off the camera" on the camera settings page and i know this option don't realy switch the cam off because the rtsp stream an snapshot works properly even the function is deactivated.

So from my point of view the feature "Switch on/off the camera" should be renamed to "Switch recording on/off".

And my question was, is it possible to plit, i don't know if it makes sense and if it is possible from technical point of view, the "Save video when a motion is detected" feature in "enable/disable motion detection" and "enable/disable recording". If boath are disabled so i can triger recording via HTTP request (record.sh).
Thanks

roleoroleo, 2021-01-17:
The meaning of the page "Camera settings" is to replicate some commands present in the app.
Whe you switch on/off a control the page sends to the system the same command sent from the app.
I tried several times but I failed to replicate the "Enable/disable motion detection".
Probably it's only a cloud option.
So it's not possible from the web page to enable or disable motion detection.

arthur0412, 2021-01-17:
ok, :o/ 
Anyway, thank you for the explanation.

closed 2021-02-11 by roleoroleo