roleoroleo/yi-hack-MStar #304: Wrong date & time

pune2001, 2021-01-12:
I'm using the latest firmware (0.3.9) and have selected & copied my timezone from the configuration page. NTP server is activated and has a valid entry.

However, even after rebooting, the configuration shows a wrong date (April 2020!) and I cannot figure out how to correct it. Everything else works just fine but I always get a totally wrong timestamp on my snapshots.

Any idea what I can do?

roleoroleo, 2021-01-13:
Try to run it manually:
`ntpd -d -n -p pool.ntp.org`

bartvdhoeven, 2021-01-24:
Thanks for responding on this issue.
I'm having the same issue here, Dome X model with latest 0.9.3 firmware.

Here is my output:

/etc/init.d # ntpd -nd -p dk.pool.ntp.org
ntpd: 'dk.pool.ntp.org' is 5.186.58.8
ntpd: sending query to 5.186.58.8
ntpd: timed out waiting for 5.186.58.8, reach 0x00, next query in 2s
ntpd: 'dk.pool.ntp.org' is 5.186.58.8
ntpd: sending query to 5.186.58.8
ntpd: timed out waiting for 5.186.58.8, reach 0x00, next query in 1s
ntpd: 'dk.pool.ntp.org' is 5.186.58.8
ntpd: sending query to 5.186.58.8
....

roleoroleo, 2021-01-25:
The connection terminates with a time out.
Is the connection filtered by a firewall?
Or try another server.

bartvdhoeven, 2021-01-29:
Thank you for responding, it was indeed a firewall issue in my case..

closed 2021-02-13 by roleoroleo