roleoroleo/yi-hack-MStar #307: Aqara Camera G2 RTSP issue

niceboygithub, 2021-01-29:
First, you are a genius and a good man to share this project and source code. God will love you.

Recently, I try to use Aqara Camera G2. And I found that it is using MStar solutions.
So I start to try to make RTSP running on it.
I already made imggrabber and h264grabber work. But when I try to enable rRTSPserver. It failed.
Can you give any hint to fix it?

Here is the debug logs
```` shell
# ./rRTSPServer

"ch0_0.h264" stream, from the file "/tmp/h264_high_fifo"
Play this stream using the URL "rtsp://10.8.28.57/ch0_0.h264"
accept()ed connection from 10.8.28.70
RTSPClientConnection[0xe1d98]::handleRequestBytes() read 127 new bytes:OPTIONS rtsp://10.8.28.57:554/ch0_0.h264 RTSP/1.0
CSeq: 2
User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)

parseRTSPRequestString() succeeded, returning cmdName "OPTIONS", urlPreSuffix "", urlSuffix "ch0_0.h264", CSeq "2", Content-Length 0, with 0 bytes following the message.
sending response: RTSP/1.0 200 OK
CSeq: 2
Date: Fri, Jan 29 2021 02:01:05 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER

RTSPClientConnection[0xe1d98]::handleRequestBytes() read 153 new bytes:DESCRIBE rtsp://10.8.28.57:554/ch0_0.h264 RTSP/1.0
CSeq: 3
User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
Accept: application/sdp

parseRTSPRequestString() succeeded, returning cmdName "DESCRIBE", urlPreSuffix "", urlSuffix "ch0_0.h264", CSeq "3", Content-Length 0, with 0 bytes following the message.
terminate called after throwing an instance of 'int'
terminate called recursively
Aborted

roleoroleo, 2021-01-29:
Unfortunately, it is not clear from the log.
First try to stream an h264 file using the testOnDemandRTSPServer demo application.

niceboygithub, 2021-01-30:
Thanks for your help and hint.

Now RTSP works!!

GOD love you.

niceboygithub, 2021-01-30:
Another question.
I found that "/proc/mstar/OMX/VMFE0" or "/proc/mstar/OMX/VMFE1" can be disabled or enabled by some commands.
Do you have any idea about it?

Thanks

roleoroleo, 2021-01-30:
If you look in the init.sh file there are many lines with the setprop command :
```
        setprop mi.vi.src 2
        setprop mi.vi.img.sub 0
        setprop mi.sys.shrink_mem 1
        setprop mstar.omx.gop.disable 1
        setprop mi.osd.gop.use 0
        setprop mi.venc.bufcnt 1
        setprop mi.venc.sub.bufcnt 1
        setprop mi.vi.bufcnt 3
        setprop mi.vi.sub.bufcnt 3
        setprop mi.vi.sub.width 640
        setprop mi.vi.sub.height 360
        setprop mi.venc.stop.flush 1
        setprop mi.video.height.force.aligned32 1
        setprop omx.mainstream.min.resolution.width 1920
        setprop omx.mainstream.min.resolution.height 1088
        setprop mstar.omx.avqe.aecgain 2
        setprop mstar.omx.avqe.aecmode 15
#       setprop mstar.omx.avqe.dbg.enable 1
#       export AVQE_DBG_PATH=/tmp/sd/
        setprop mi.vi.yuvmon.pat /home/ms
```

I don't know if that's what you mean.

niceboygithub, 2021-01-30:
Thanks, you give a good hint.

Here are G2's settings.
```
[Video.0.H264VBR.ChangePos]: [80]
[Video.0.H264VBR.Gop]: [40]
[Video.0.H264VBR.IPQPDelta]: [0]
[Video.0.H264VBR.MaxBitrate]: [524288]
[Video.0.H264VBR.MaxIQp]: [51]
[Video.0.H264VBR.MaxPQp]: [51]
[Video.0.H264VBR.MinIQp]: [10]
[Video.0.H264VBR.MinPQp]: [10]
[Video.0.H264VBR.StatTime]: [0]
[Video.0.H265VBR.ChangePos]: [80]
[Video.0.H265VBR.Gop]: [40]
[Video.0.H265VBR.IPQPDelta]: [0]
[Video.0.H265VBR.MaxBitrate]: [1048576]
[Video.0.H265VBR.MaxIQp]: [51]
[Video.0.H265VBR.MaxPQp]: [51]
[Video.0.H265VBR.MinIQp]: [10]
[Video.0.H265VBR.MinPQp]: [10]
[Video.0.H265VBR.StatTime]: [0]
[Video.0.RC.RcVersion]: [0]
[Video.0.RC.RowQpDelta]: [0]
[Video.0.RC.method]: [H264_VBR]
[Video.1.H265VBR.ChangePos]: [80]
[Video.1.H265VBR.Gop]: [40]
[Video.1.H265VBR.IPQPDelta]: [0]
[Video.1.H265VBR.MaxBitrate]: [786432]
[Video.1.H265VBR.MaxIQp]: [51]
[Video.1.H265VBR.MaxPQp]: [51]
[Video.1.H265VBR.MinIQp]: [10]
[Video.1.H265VBR.MinPQp]: [10]
[Video.1.H265VBR.StatTime]: [0]
[Video.1.RC.RcVersion]: [0]
[Video.1.RC.RowQpDelta]: [0]
[Video.1.RC.method]: [H265_VBR]
[init.svc.rcs]: [running]
[init.svc.ueventd]: [running]
[mi.debug.level]: [8]
[mi.debug.proc.ADEC.ready]: [1]
[mi.debug.proc.AENC.ready]: [1]
[mi.debug.proc.AI.ready]: [1]
[mi.debug.proc.AO.ready]: [1]
[mi.debug.proc.MI.ready]: [1]
[mi.debug.proc.OSD.ready]: [1]
[mi.debug.proc.VENC.ready]: [1]
[mi.debug.proc.VI.ready]: [1]
[mi.sys.shrink_mem]: [1]
[mi.venc.bufcnt]: [2]
[mi.venc.sub.bufcnt]: [2]
[mi.vi.bufcnt]: [3]
[mi.vi.src]: [2]
[mi.vi.sub.bufcnt]: [2]
[mi.vi.sub.height]: [720]
[mi.vi.sub.width]: [1280]
[mstar.omx.gop.disable]: [1]
[mstar.omx.log.level]: [0]
[persist.sys.usb.config]: [adb]
[ro.allow.mock.location]: [0]
[ro.baseband]: [unknown]
[ro.bootloader]: [unknown]
[ro.bootmode]: [unknown]
[ro.debuggable]: [1]
[ro.factorytest]: [0]
[ro.hardware]: [mstarinfinity3(flatteneddevicet]
[ro.revision]: [0]
[ro.secure]: [1]
[ro.serialno]: []
[ro.system_rcs]: [1]
```

niceboygithub, 2021-02-05:
Another question again

How do you know the input is “ YUV (YUV 420SP NV12”?

thansk

roleoroleo, 2021-02-05:
Reverse engineering and some good tools: https://rawpixels.net/

nutsben, 2021-03-06:
> Thanks for your help and hint.
> 
> Now RTSP works!!
> 
> GOD love you.

Hi,
did you succeed in enabling rtsp  on camera g2?
i am very interested to know how i can also do it

Thanks a lot

niceboygithub, 2021-03-07:
> > Thanks for your help and hint.
> > Now RTSP works!!
> > GOD love you.
> 
> Hi,
> did you succeed in enabling rtsp on camera g2?
> i am very interested to know how i can also do it
> 
> Thanks a lot

For G2, RTSP server only work low resolution and firmware version 3.4.6 and 3.5.7
In the latest version, it changed to h265. I did not found the solution to parsing the yuv format.

roleoroleo, 2021-03-07:
> In the latest version, it changed to h265. I did not found the solution to parsing the yuv format.

Check my project:
https://github.com/roleoroleo/yi-hack-Allwinner-v2
r30gb works with h265.

closed 2021-05-08 by roleoroleo