roleoroleo/yi-hack-MStar #308: Wie finde ich die richtige Firmware

RalfEngelberth, 2021-01-29:
Hallo, ich habe eine Floureon Camera Outdoor XY-R9820-G1 mit der Firmware 7.1.00 und möchte RTSP nutzen.
Ich habe schon alles versucht. Ich bekomme die neue Firmware nicht auf die Cam
Kann mir einer einen Tipp geben, welche Firmware für mein Cam die richtige ist?
Eine Telnet-Verbindung bekomme ich auch nicht aufgebaut.

Was mache ich falsch bzw. welche Firmware ist die richtige?

MD-V, 2021-01-30:
Hello Ralf,

please write your next posts in English so that the majority on the platform can understand your post.

This firmware is specifically designed for cameras with an MStar SoC. I can't find the any information about your cameras SoC online.

Do you have an QR Code somewhere on your camera? If yes there should be text under the QR code. If the first four letters of the text appear in the following table you have identified the correct firmware: https://github.com/roleoroleo/yi-hack-MStar#supported-cameras. 

If not your camera is probably not supported by this firmware.

RalfEngelberth, 2021-01-30:
Thanks for the fast respond.
The first 4 letters are A001 ....
Do I have other options to get the cam on RTSP?

MD-V, 2021-01-30:
@RalfEngelberth Telnet is not working?

RalfEngelberth, 2021-01-30:
@MD-V 
No

RalfEngelberth, 2021-02-03:
Hello, I opened the camera and took a picture of the circuit board. Can someone give me a tip on how to hack or open RTSP?

[image]

[image]

MD-V, 2021-02-05:
As this [URL](https://www.gadgetvictims.com/2019/11/digoo-dg-k2-ive-seen-this-face-before.html) says your camera seems to be based on the RTS3903N chipset. For this chipset you have to look at https://github.com/TheCrypt0/yi-hack-v4/issues/29 

There is currently no custom firmware available for your camera.
I think you can close this issue.

RalfEngelberth, 2021-02-05:
thanks for the reply

closed 2021-02-05 by RalfEngelberth