roleoroleo/yi-hack-MStar #309: maybe ... a silly question ...

maurofilippo, 2021-02-01:
hello, first of all congratulations for everything that has been done!
Last night I tried to update the original fw of my Yi 1080p home and it worked and I'm super happy!
Now we come to the probable silly question ... Before updating the firmware I had already paired the ip cam with the home wireless network, so after updating the firmware she reconnected to the network .... but if instead if you have done the reset and then lost the wifi connection, the camera with the new fw is always programmed by pairing with the mobile phone?
This question why should I change the ssid address and password and I don't know how to do it except resetting it and re-pairing ...
thank you very much for an answer.
good day
Mauro

roleoroleo, 2021-02-01:
If you disable the check "Disable cloud" you should be able to resync with the app.

maurofilippo, 2021-02-01:
thanks for the answer, sorry if I ask you again but I'm afraid of losing the connection and then not being able to reach it anymore.
So you are telling me to remove the flag from the direct connection to the ip cam ie http: // ip: 8080, configuration, disable cloud = active?
done this then what should I do, reconnect with the app and generate the qrcode and then pair again with the smartphone?
the firmware then did not modify that part there, the cam on behaves as with the original fw?

sorry again for the trouble and also for the English (I'm using a translator)

Hi

roleoroleo, 2021-02-01:
You have 2 ways:
- restore the original fw
- try to set wifi credentials with this script: https://github.com/roleoroleo/yi-hack-MStar/wiki/Change-WiFi-credentials

roleoroleo, 2021-02-13:
News about your issue?

maurofilippo, 2021-02-14:
Hi! how kind to ask me!
So, I first tried with the file you told me to download. I put SSID and PASSWORD without quotes. I turned off the cam, inserted the micro sd, the yellow light came on, then the blue one repeatedly ... after a while (1 or 2 minutes) I decided to turn it off. I removed the micro sd and when it turned on nothing changed, that is, it connected to the usual wifi network ... so it didn't work ....
Plan B: I reset the cam (very afraid of doing damage ...) and simply, as you told me, I reprogrammed with the correct wifi .... and it worked!
What I would like, however, is not to be tied to the app forever but to be able to manage the cam alone .... anyway that's okay, I still thank you for the super job you did.
Ps: maybe I say a triviality, I would just like to point out to the community that if you disable the cloud function from the web page of the ip cam, the app will no longer communicate with the cam and therefore if it is reset, then to set the wifi son pains because it does not paring anymore ...... so ..... solution ..... reinstall the firmware which fortunately has the default cloud service active.
thanks again for the intetessamento.
Hi :-)

closed 2021-02-19 by roleoroleo