roleoroleo/yi-hack-MStar #314: MQTT _motion_files_ messages timing

1300371, 2021-02-09:
When the _upload features_ is enabled in **Motion Events** feature and files has been generated, I receive the _mqtt_ message concerning the motion files a long time **before** the files has been completely uploaded.

Is it possible to send _motion_files_ mqtt message after the upload process is done ?

roleoroleo, 2021-02-10:
It's not possible.
Motion file event is generated after the dameon receives a stop motion message.
And ftp upload can start only when files are generated from the yi daemon, so after motion file event.

1300371, 2021-02-10:
Alright.
Is it possible to create a new mqtt message when the upload is finished.

To explain my situation:
- I upload videos on remote FTP server
- I provide after links to these files in HA on _motion_files_ message

Unfortunately, as the _motion_files_ arrives before the upload finished, I have to wait until my link really works.

roleoroleo, 2021-02-10:
Check if you can reach the same result using web services.
https://github.com/roleoroleo/yi-hack-MStar/wiki/Web-services-description

1300371, 2021-02-10:
The problem with _WS_ is I must pull the info. Instead of mqtt that provides subscriptions.
I can understand it can't be achieved: no problem.

BTW, thank you for your time. 👍

closed 2021-05-08 by roleoroleo