roleoroleo/yi-hack-MStar #316: Expose settings through MQTT

MoiZie, 2021-02-12:
Maybe I'm overlooking things, but is there a way to disable video recording through MQTT? Use case: when I'm home, I don't need recordings of my rooms, nice privacy feature. It also saves wear on the SD card and reduces the WiFi usage.

MoiZie, 2021-02-12:
For an example of MQTT control integration see the Dafang Hacks:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/firmware_mod/scripts/mqtt-control.sh

roleoroleo, 2021-02-12:
At the moment it's possible only through web services.
But I could do this.

Smosia, 2021-05-09:
Any news about this feature?
I found "Topics Prefix for remote configuration" in MQTT web page. 
What is it?

roleoroleo, 2021-05-09:
Yes, check this pr:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/pull/114

Smosia, 2021-05-09:
Thank you. I will test 2 method.

Maybe you could add that info to Readme.md?
It would be helpful for new users.

roleoroleo, 2021-05-09:
Yes.

lesensei, 2021-05-20:
I'm sorry I'm feeling pretty dumb, but I don't get it :-(

What does the "file_to_update" reference points to ? Where can I find a list of these files and the parameters they each host ?

roleoroleo, 2021-05-21:
Read the configuration files with the web service:
```
http://IP-CAM:8080/cgi-bin/get_configs.sh?conf=system
http://IP-CAM:8080/cgi-bin/get_configs.sh?conf=mqtt
```

system is the content of the file system.conf
mqtt is the content of the file mqtt.conf
So "file_to_update" can be system, mqtt, etc...

closed 2024-05-09 by github-actions[bot]