roleoroleo/yi-hack-MStar #318: [Feature request] - WiFi discovery through the Web Gui

robustini, 2021-02-27:
It would be interesting to be able to WiFi discovery through the Web Gui, in order to associate the cam to a new network without necessarily resetting it or manually acting on the file system.

roleoroleo, 2021-03-03:
I could send you a beta release to test, if you want.

robustini, 2021-03-03:
> 
> 
> I could send you a beta release to test, if you want.

Sure, thanks!

roleoroleo, 2021-03-03:
What's your model suffix?

robustini, 2021-03-04:
BFUS, so h201c.

roleoroleo, 2021-03-04:
Here it is:
[h201c_0.3.9.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6083907/h201c_0.3.9.tar.gz)

robustini, 2021-03-04:
Thanks @roleoroleo!
All work fine except saving the credentials.
At the first error attempt, it always works on the following ones.
After a reboot same thing, first attempt with error.

[image]

[image]

roleoroleo, 2021-03-04:
I don't understand...
I have never seen this error in all the tests I have done.

roleoroleo, 2021-03-04:
There is an error server side.
Could you try to run a post action to this page

http://IP:8080/cgi-bin/wifi.sh?action=save

with the following content?
``
{
"WIFI_ESSID":"essid",
"WIFI_PASSWORD":"pwd",
"WIFI_PASSWORD2":"pwd"
}

robustini, 2021-03-05:
> 
> 
> I don't understand...
> I have never seen this error in all the tests I have done.

I tried again now, always on the first save attempt after a reboot it gives me that problem, on the second and subsequent ones it doesn't.

roleoroleo, 2021-03-05:
I tried again with my y203c and I have no problems.

robustini, 2021-03-05:
I repeat, it is a relative problem, so much in reality it saves the settings, what I would not like is that someone else can report it to you in the future. 
I don't think it's my BFUS's fault (i hope).

roleoroleo, 2021-03-05:
I will release this feature in the next version.
If there is a problem we will see it.
:)

robustini, 2021-03-06:
@roleoroleo I went back to 0.3.9 because with this firmware and the same saved settings the rtsp doesn't work, don't ask me why.

roleoroleo, 2021-03-06:
Could you please test this version?
RTSP server should be fixed.
[h201c_0.3.9.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6095500/h201c_0.3.9.tar.gz)

robustini, 2021-03-07:
> 
> 
> Could you please test this version?
> RTSP server should be fixed.
> [h201c_0.3.9.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6095500/h201c_0.3.9.tar.gz)

Sorry, same problem.
I double checked, the configuration is all ok.

roleoroleo, 2021-03-07:
What's the problem?
It starts and crashes?
It doesnt' start at all?

robustini, 2021-03-07:
It cannot find the server on port 554.

roleoroleo, 2021-03-07:
Did you reconfigure the cam after the update?

robustini, 2021-03-07:
No, I imported the previously saved configuration, but I checked that everything was as before and it is.

roleoroleo, 2021-03-07:
Ok, I will try again.

robustini, 2021-03-09:
> 
> 
> Ok, I will try again.

If I can help you I can check if the rtsp server is running correctly after a reboot.
How can I check it from the shell?

steef84, 2021-03-09:
@roleoroleo is it possible for me to beta test y203c. Got to reconfigure a cam so update is nice to have.

robustini, 2021-03-18:
No news?

roleoroleo, 2021-03-18:
[y203c_0.3.9.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6164421/y203c_0.3.9.tar.gz)
[h201c_0.3.9.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6164425/h201c_0.3.9.tar.gz)

robustini, 2021-03-18:
> 
> 
> [y203c_0.3.9.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6164421/y203c_0.3.9.tar.gz)
> [h201c_0.3.9.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6164425/h201c_0.3.9.tar.gz)

Perfect, RTSP stream is back!

steef84, 2021-03-23:
Nice addon to configure wifi via webinterface! Nice possibility to add is function for adding hidden networks also via the webinterface. Now it only scans visible networks.

roleoroleo, 2021-03-23:
What do you mean?
A text box where enter manually a ssid name?

steef84, 2021-03-24:
Yes, exactly!

closed 2024-05-09 by github-actions[bot]