roleoroleo/yi-hack-MStar #320: 4FUS bootlooping after clicking reboot

leandroarthur, 2021-03-04:
Other details:
- Sd card size 128gb (119gb in yi app)
- Current usage 23.4 of 119gb
- Tried using 5V 2A power adapter, still bootlooping
- No bootlooping if sdcard is removed
- it bootloops after light indicator turns steady blue for 5-6s

Help needed @roleoroleo 

Is this a problem of
- defective sd card
- camera memory issue
- 128gb sd incompatibility with yi-hack

Thank you!

leandroarthur, 2021-03-04:
Surprisingly, the camera is no longer bootlooping, just minutes after I wrote about the issue. Its current power adapter is 5V 2A.
But i am still uncomfortable to reboot the device again 🤣

roleoroleo, 2021-03-04:
Mmmmmh.
Maybe a power issue (cable?) or an incompatible sd card.

leandroarthur, 2021-04-03:
Hi @roleoroleo !
I replaced my 128gb sd with a new one. Using a new power cable too plus the 5v 2a adapter.
But still getting bootloops.

At 13.2/119gb usage, I disconnected power and restarted the camera, no bootlooping occurred.
At 27.3/119gb usage, I disconnected power and restarted it again, bootlooping happened.
Disconnected power, removed sd, deleted all its contents, reinserted sd back to the camera, and restarted the camera, no bootlooping occurred.

roleoroleo, 2021-04-03:
Maybe a problem with hd sd.
Try another sd card.

leandroarthur, 2021-04-03:
@roleoroleo !!
I think I found the cause.
Found out that Yi Home app formatted my sd card with 32KB allocation unit size.
I reformatted the sd with 64KB a.u.s. and put some good amount of old footages into it. No longer bootlooping.

closed 2021-04-06 by roleoroleo

leandroarthur, 2021-04-10:
@roleoroleo man,
Experiencing the bootloop issue again when sd usage hits over 64gb.
Not sure if the official yi firmware revision [4.2.0.0H_20190904162 ](https://forum.yitechnology.com/t/yi-home-camera-3-firmware-revision/223)is causing this
As a temporary workaround, I created a 64gb fat32 primary partition within the 128gb space.

roleoroleo, 2021-04-10:
This is not a dump but an official fw upgrade:
[4.2.0.0F_201905241332home_y25m.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6289378/4.2.0.0F_201905241332home_y25m.tar.gz)
But I don't know if downgrade works.

leandroarthur, 2021-04-10:
should i rename 4.2.0.0F_201905241332home_y25m.tar.gz as home_y25m
or this downgrade requires serial port connection?

roleoroleo, 2021-04-10:
Sorry, wrong file.
Unzip this in the sd card:
[4.2.0.0F_201905241332home_y25m.zip](https://github.com/roleoroleo/yi-hack-MStar/files/6289885/4.2.0.0F_201905241332home_y25m.zip)

leandroarthur, 2021-04-10:
thank you for this. but i was unsuccessful.

what i did:
1. currently using mstar 0.3.9
2. downgraded base firmware from 4.2.0.0H_201909041620 to 4.2.0.0F_201905241332
3. restarted yi cam without sd while filling sd with 110 gb of footage
4. reinserted sd back to camera. infinite bootloops.
5. removed sd. put mstar 0.40 firmware files to sd.
6. reinserted sd back to camera (still with 110gb of footage). 3 bootloops then cam went ok. mstar now 0.4.0. base firmware gets auto-updated too to 4.2.0.0H_201909041620
7. disconnected camera's power for some seconds.
8. restarted camera with the unremoved sd. infinite bootloops.

maybe i'll stick with my temp workaround for now

leandroarthur, 2021-04-24:
@roleoroleo, just reporting back because I solved my issue (finally).

1. Formatted my SD card using [SD Association's official formatter](https://www.sdcard.org/downloads/formatter/). It formatted the card as exFAT with 128KB allocation unit size.
2. Then reformatted the card using a [FAT32 formatter by Ridgecrop](http://ridgecrop.co.uk/index.htm?guiformat.htm). The formatter provides lower allocation unit size options — I set mine to 8KB (8192 bytes).
3. I put old videos to the card (about 100GB+ in total size). Reinserted it to the cam. Restarted the cam. Bootlooping gone.

Thank you.