roleoroleo/yi-hack-MStar #326: Reset password for rtsp server and reflashing of firmware

nupnup-goddamn, 2021-03-18:
Hi, 
I recently set the username and password for rtsp server but apperently i now cant access my rtsp stream. I have reformatted my sdcard to FAT32 and copy the required files to the root folder of the sdcard. 
However, i cant seem to get the firmware of the camera to flash the sdcard. 
I followed the steps from the repo:
1) remove power supply of the camera
2) insert sdcard
3) plug the power supply back in

However, i dont get the yellow flashing light. 

Did i do the reflashing correctly? will reflashing the fimrware help me to reset the password setting of the rtsp as well?

roleoroleo, 2021-03-18:
When you already flashed a version, you can't reflash it.
Because the system save the md5 sum of the image.
You have to flash a previous version and then upgrade again.

nupnup-goddamn, 2021-03-19:
Thanks it works! may i also ask how can i improve the framerate of the rtsp streaming? which script can i edit?

roleoroleo, 2021-03-19:
Sorry, it's not possible to change framerate.

nupnup-goddamn, 2021-03-20:
I see. Thanks anyway for resolving my re-flashing issue! Cheers

closed 2021-03-20 by nupnup-goddamn