roleoroleo/yi-hack-MStar #331: Supported Camera Models

marcel89l, 2021-03-22:
Hey together and thanks to @roleoroleo  for your awesome work!
Just received my yi 1080p Home camera and choosed to flash this hack.
Unfortunately i wasn't sure which version i had to flash, so i unscrew the cam to look what serial number and chipset is inside it and was hoping i find with this info some comment which version suits my camera. 

Seems to be a quite new version, i haven't find any comment which version i have to choose. Tried my luck with mstar and also allwinner, which both doesn't breake my cam (i was lucky) but both also doesn't work (no hack happens, camera is booting normal).
Is this probably the cause that this camera version isn't supported (yet)? Or should it work and i'm doing something wrong...?

Not sure which infos are useful, so i provide everything i can find from the front-/ and backside of the cam
Front: 
y29ga_mb_v2.1
2020/04/07
realtek 8189ftv k91ky31 gkj9c
winband 25064jvsiq 2026

Back:
qg2101a l1041aa63t2 
 
[image]
[image]

roleoroleo, 2021-03-22:
https://github.com/roleoroleo/yi-hack-Allwinner-v2

marcel89l, 2021-03-22:
Thanks, awesome, works out of the box :)!
I downloaded the y21ga version, copied it to the sd card and now my cam boots with hack activated! Thank you very much @roleoroleo for this fast reply!!

closed 2021-03-22 by marcel89l