roleoroleo/yi-hack-MStar #332: web-dashboard empty after upgrading from 0.3.9 to 0.4.0, YI Dome (H201C)

pejotigrek, 2021-03-22:
**camera model:** YI Dome 1080P 360 PTZ (H201C)
**Yi-Hack firmware:** updated from 0.3.9 to 0.4.0

I'm using your firmware on my two cameras - Yi Home & Yi Dome. web-dashboard showed me that there's an update [which I confirmed by checking the github] so I decided to make an automatic update via web. clicked the button and after a while - camera booted with, I assume, newer firmware, but when I log into the web-dashboard, everything is empty, see the screens:

[image][image][image][image]

I know that every setting I had works fine, as my HomeAssistant instance, tinyCam PRO etc. connects without any issues, old streaming links work fine, PTZ links work fine and MQTT seems to be working too. I'm just concerned, that I lost the ability to control my cam via web, which would impact future semi-automatic firmware upgrades too.

is there anything I should do to get the dashboard working again, or the only way is to try factory reset & use your firmware "from scratch"?

right now I'm not upgrading the Yi Home camera, as I don't want to loose those options too [this camera is outdoor, closed in weather-proof case, little bit too difficult to get the sd card out for firmware flashing in case of troubles ;)]

roleoroleo, 2021-03-22:
Try to clear the cache of your browser.
CTRL+F5

pejotigrek, 2021-03-22:
I did tried that in the first place, unfortunately with no effect.
also tried to clear site data within web developer tools in Chrome, and also tried the "clear cache and reload page" option which is available when right-clicking the reload button with developer tools active.

in addition I tried that on few separate browsers [at home, at work via VPN and directly with port-forwarding on my router, and also on mobile]. no effect, page stays "clear" everywhere.

if it can help in any way - while being in developer tools I've checked the console and here's what I got after force refreshed reload:
[image]

roleoroleo, 2021-03-22:
There is a problem in your cgi-bin folder.
Check the content of this folder through ssh.

pejotigrek, 2021-03-22:
ok, checked - these are the contents:

```
root@XXX.XXX.XXX.XXX's password:

                   _     _           _
               _ _|_|___| |_ ___ ___| |_
              | | | |___|   | .'|  _| '_|
              |_  |_|   |_|_|__,|___|_,_|
              |___|
 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.3.9
 -----------------------------------------------------

/home/yi-hack # ls
bin           etc           lib           model_suffix  sbin          script        version       www
/home/yi-hack # ls www
camera_settings.sh       eventsdirdel.sh          get_configs.sh           index.html.gz            pages                    record                   service.sh
cgi-bin                  eventsfile.sh            getlastrecordedvideo.sh  js                       preset.sh                record.sh                set_configs.sh
css                      eventsfiledel.sh         hostname.js              links.sh                 ptz.sh                   reset.sh                 snapshot.sh
eventsdir.sh             fw_upgrade.sh            images                   load.sh                  reboot.sh                save.sh                  status.json
/home/yi-hack # ls www/cgi-bin
TZ                camera.conf       hostname          mqttv4.conf       passwd            proxychains.conf  system.conf
/home/yi-hack #
```

hmm, and it looks like the update didn't went through, right? still 0.3.9 in the header..
is it enough if I save the update files on the sdcard and reboot the camera?

roleoroleo, 2021-03-23:
Wow...
Something went wrong with the update.
Your cgi-bin folder is completely wrong.
You need to apply again the hack.

pejotigrek, 2021-03-23:
yeah, I thought so - it definitely looked weird ;)
I will try to apply the hack from scratch then. thanks!

closed 2021-03-23 by pejotigrek