roleoroleo/yi-hack-MStar #333: MQTT login issue

Rushmed, 2021-03-22:
Since the updtae from 0.3.9 to 0.4.0 I get MQTT login error.
`
(7343) Client [856] has invalid password(undefined) or username(undefined)
`
On 0.3.9 it was working fine.
I checked login and password at the webinterface and in mqttv4.conf. Both are correct. Also checked the credentials in my broker.
Tried changing password on cam and broker without success.
Also did a reset of the hack and of course several reboots.

Got the same issue on latest Allwinner.

roleoroleo, 2021-03-22:
I will check it.

Rushmed, 2021-03-22:
Thats curious. Did something change MQTT guidelines? Maybe no capitals or something lik that?

roleoroleo, 2021-03-22:
The only change related to mqttv4 is https://github.com/roleoroleo/yi-hack-MStar/commit/b402b08b26847fdf98e4385291e0dcd2d66e1f20

It shouldn't be a critical change.

Rushmed, 2021-03-22:
Yes, with downgrade to 0.3.9 it works as expected.

majkers, 2021-03-22:
Me too have problem with mqtt. I can login but both my camera keep disconnecting
1616445707: New connection from 192.168.1.11 on port 1883.
1616445707: Socket error on client <unknown>, disconnecting.
1616445708: New connection from 192.168.1.10 on port 1883.
1616445708: Socket error on client <unknown>, disconnecting.

roleoroleo, 2021-03-22:
Check the name of the client.
If you try to use the same name on 2 different client you have this error.

majkers, 2021-03-22:
Nothing changes on that sińce 0.3.9. Client ids are different

majkers, 2021-03-22:
1616446521: New client connected from 192.168.1.11 as xxxDome2 (p2, c1, k120, u'mqtt_user').
1616446521: New client connected from 192.168.1.10 as xxxHome1 (p2, c1, k120, u'mqtt_user').
1616446521: New connection from 192.168.1.10 on port 1883.
1616446521: Socket error on client <unknown>, disconnecting.
1616446523: New connection from 192.168.1.11 on port 1883.
1616446523: Socket error on client <unknown>, disconnecting.

roleoroleo, 2021-03-22:
Are you using authentication?
Could you share your mqtt config?

roleoroleo, 2021-03-22:
Another question?
Are you sure the this error comes from mqttv4 program?
Or it comes from the new mqtt-config program?
Check if motion events arrive to the mqtt server.

majkers, 2021-03-23:
I am not sure it comes from mqttv4 . This is part of mosquitto's Home Assistant log. Motion detection seems to work ok but these IP adresses are asssigned to my cameras with your firmware. I have other two with older firmware and witohut that problemm and lots of stuff on zigbee2mqtt. No isssues there. Here is my config (I changed some vulnerable data):

[image]
[image]
[image]
[image]

majkers, 2021-03-23:
And this is my log form the morning:

`1616481155: New connection from 192.168.1.11 on port 1883.
1616481155: Socket error on client <unknown>, disconnecting.
1616481157: New connection from 192.168.1.10 on port 1883.
1616481157: Socket error on client <unknown>, disconnecting.
1616481160: New connection from 192.168.1.11 on port 1883.
1616481160: Socket error on client <unknown>, disconnecting.
1616481162: New connection from 192.168.1.10 on port 1883.
1616481162: Socket error on client <unknown>, disconnecting.
1616481165: New connection from 192.168.1.11 on port 1883.
1616481165: Socket error on client <unknown>, disconnecting.
1616481167: New connection from 192.168.1.10 on port 1883.
1616481167: Socket error on client <unknown>, disconnecting.
1616481170: New connection from 192.168.1.11 on port 1883.
1616481170: Socket error on client <unknown>, disconnecting.
1616481172: New connection from 192.168.1.10 on port 1883.
1616481172: Socket error on client <unknown>, disconnecting.
1616481175: New connection from 192.168.1.11 on port 1883.
1616481175: Socket error on client <unknown>, disconnecting.
1616481177: New connection from 192.168.1.10 on port 1883.
1616481177: Socket error on client <unknown>, disconnecting.
1616481180: New connection from 192.168.1.11 on port 1883.
1616481180: Socket error on client <unknown>, disconnecting.
1616481182: New connection from 192.168.1.10 on port 1883.
1616481182: Socket error on client <unknown>, disconnecting.
1616481185: New connection from 192.168.1.11 on port 1883.
1616481185: Socket error on client <unknown>, disconnecting.
1616481187: New connection from 192.168.1.10 on port 1883.
1616481187: Socket error on client <unknown>, disconnecting.
1616481190: New connection from 192.168.1.11 on port 1883.
1616481190: Socket error on client <unknown>, disconnecting.
1616481192: New connection from 192.168.1.10 on port 1883.
1616481192: Socket error on client <unknown>, disconnecting.
1616481195: New connection from 192.168.1.11 on port 1883.
1616481195: Socket error on client <unknown>, disconnecting.
1616481197: New connection from 192.168.1.10 on port 1883.
1616481197: Socket error on client <unknown>, disconnecting.
1616481200: New connection from 192.168.1.11 on port 1883.
1616481200: Socket error on client <unknown>, disconnecting.
1616481202: New connection from 192.168.1.10 on port 1883.
1616481202: Socket error on client <unknown>, disconnecting.
1616481205: New connection from 192.168.1.11 on port 1883.
1616481205: Socket error on client <unknown>, disconnecting.
1616481207: New connection from 192.168.1.10 on port 1883.
1616481207: Socket error on client <unknown>, disconnecting.
1616481210: New connection from 192.168.1.11 on port 1883.
1616481210: Socket error on client <unknown>, disconnecting.
1616481212: New connection from 192.168.1.10 on port 1883.
1616481212: Socket error on client <unknown>, disconnecting.
1616481215: New connection from 192.168.1.11 on port 1883.
1616481215: Socket error on client <unknown>, disconnecting.
1616481218: New connection from 192.168.1.10 on port 1883.
1616481218: Socket error on client <unknown>, disconnecting.
1616481220: New connection from 192.168.1.11 on port 1883.
1616481220: Socket error on client <unknown>, disconnecting.
1616481223: New connection from 192.168.1.10 on port 1883.
1616481223: Socket error on client <unknown>, disconnecting.
1616481225: New connection from 192.168.1.11 on port 1883.
1616481225: Socket error on client <unknown>, disconnecting.
1616481228: New connection from 192.168.1.10 on port 1883.
1616481228: Socket error on client <unknown>, disconnecting.
1616481230: New connection from 192.168.1.11 on port 1883.
1616481230: Socket error on client <unknown>, disconnecting.
1616481233: New connection from 192.168.1.10 on port 1883.
1616481233: Socket error on client <unknown>, disconnecting.
1616481235: New connection from 192.168.1.11 on port 1883.
1616481235: Socket error on client <unknown>, disconnecting.
1616481238: New connection from 192.168.1.10 on port 1883.
1616481238: Socket error on client <unknown>, disconnecting.
[... 7 lines cut ...]
1616481248: Socket error on client <unknown>, disconnecting.
1616481250: New connection from 192.168.1.11 on port 1883.
1616481250: Socket error on client <unknown>, disconnecting.
1616481253: New connection from 192.168.1.10 on port 1883.
1616481253: Socket error on client <unknown>, disconnecting.
1616481255: New connection from 192.168.1.11 on port 1883.
1616481255: Socket error on client <unknown>, disconnecting.
1616481258: New connection from 192.168.1.10 on port 1883.
1616481258: Socket error on client <unknown>, disconnecting.
1616481260: New connection from 192.168.1.11 on port 1883.
1616481260: Socket error on client <unknown>, disconnecting.
1616481263: New connection from 192.168.1.10 on port 1883.
1616481263: Socket error on client <unknown>, disconnecting.
1616481265: New connection from 192.168.1.11 on port 1883.
1616481265: Socket error on client <unknown>, disconnecting.
1616481268: New connection from 192.168.1.10 on port 1883.
1616481268: Socket error on client <unknown>, disconnecting.
1616481270: New connection from 192.168.1.11 on port 1883.
1616481270: Socket error on client <unknown>, disconnecting.
1616481273: New connection from 192.168.1.10 on port 1883.
1616481273: Socket error on client <unknown>, disconnecting.
1616481275: New connection from 192.168.1.11 on port 1883.
1616481275: Socket error on client <unknown>, disconnecting.
1616481278: New connection from 192.168.1.10 on port 1883.
1616481278: Socket error on client <unknown>, disconnecting.`

majkers, 2021-03-23:
If it is connected (motion events working) why it is trying to establish so many new connections?

roleoroleo, 2021-03-23:
With the last version I added a new mqtt program useful to configure the cam: mqtt-config.
Probably the error depends on this feature.
Try to leave blank the text box "Topics Prefix for remote configuration", mqtt-config should exit before to try the connection to the mqtt server.

majkers, 2021-03-23:
OK I will give it a try...and get back with results

majkers, 2021-03-23:
Yeah. That was it. Thanks!

Rushmed, 2021-03-23:
That does not help for my issue.

roleoroleo, 2021-03-23:
Is mqtt-config running?
Try a ps command.

Rushmed, 2021-03-23:
No change.
Despite these annoxing messages the mqtt functionality is working. I get state chnages for motiond detection and so on.

I am facing this issue using ioBroker. Also tried different adapters, mqtt and sonoff. wit the same type of issue.

Rushmed, 2021-04-28:
Solved with version 0.4.1.
Thanks!

closed 2021-04-28 by Rushmed