roleoroleo/yi-hack-MStar #335: Can't upgrade Dome X 0.3.8 to 0.3.9 or 0.4.0

Cho0Cho0, 2021-03-24:
Hi,

I'm running Dome X on 0.3.8 and wasn't able to upgrade it to 0.3.9 and 0.4.0.

I've tried to format the SD card and put only 2 files there: _home_y30_ and _sys_y30_, then power cycle. Upgrade is not happenning. Then I also reset the camera with the same card inserted, still no luck.

The card is working, cause it's the same one I used for initial 0.3.8 flash. And I can see that files are read from it (see the details below).

Upgrade should be triggered if either file is available: `/tmp/sd/home_y30**m**` or `/home/home_y30**m**`. The files shipped with 0.4.0 do not have that suffix and are named "(home|sys)_y30". If I copy files with that suffix into SD with manually adding **m** suffix and reboot the camera, I can see `home_y30m**.done**` file later. But web page still tells me I'm on 0.3.8.

Upgrade via web UI doesn't work. It doesn't show available 0.4.0 (however I recall 0.3.9 appeared some time before). And if I click upgrade it ends up with: `Unable to download firmware file.`

Is there any log? Or URL I should be able to check from the cam ssh? Much appreciated.

roleoroleo, 2021-03-24:
The upgrade procedure looks for files without "m" suffix.
Files with "m" suffix are used for a different type of upgrade. If you rename the files, there is a risk to brick the cam.
Please take a snapshot of your status page.

Cho0Cho0, 2021-03-24:
Sure

[image]

roleoroleo, 2021-03-24:
The online upgrade procedure probably doesn't work because the root file system is full (know bug).
But the upgrade through the sd card must work.
Please try again with a different SD card.

Cho0Cho0, 2021-03-26:
Yes, changing the SD card helped. Thank you!

closed 2021-03-26 by Cho0Cho0