roleoroleo/yi-hack-MStar #337: Continuos MQTT topic  'status'=online

AirmakZ, 2021-03-25:
Hi,

I have a strange behavior on one of my Yi1080p cams.
When I activate MQTT the camera is starting ti send approximately 1 time per second the topic STATUS = online.
It's not fixed time, but around 1 and 1.5 second.
I noticed it after 0.4.0 update, but I downgrade it to 0.3.9 and... same.... So I cannot be sure it happened after 0.4 upgrade.

I made already other test other than downgrade, as try to reset and repairing using official Yi app.
Also I reset all configurations "Reset to yi-hack default", and once I activate MQTT non stop messages starts again.

I could see a very strange thing.... If I update the topic name for status (changing it from 'status' to 'statusx') after restart both topics are sent in parallel !!! Similar timing but not at the same time. Like two instances of the code running because the gap between same messages change to 2 seconds

I don't know if it can help, but 'motion' topic is always 'motion_start'

Do you have an idea about what could be the root cause of this or even just how reset some part of the camera to try to solve it?

If do you think some additional data could help you, please ask me and I will try to obtain it !

Thanks in advanced !!!

n1kDon, 2021-04-17:
Stupid question but are you sure that you changed your "client id" and "Topics prefix" in the mqtt settings to an unique id?

roleoroleo, 2021-04-17:
Check this issue:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/275

AirmakZ, 2021-04-17:
Oh god... it's not stupid question.... :-o
It seems I mistook que config file while restoring configuration and CAM1 and CAM3 where using the same ID and topic prefix !!!!
Thanks guys for your support... What was stupid was my mistake.... and don't notice it....  :\

closed 2021-04-17 by AirmakZ