roleoroleo/yi-hack-MStar #338: Issues with 0.4.0, losing connection with Synology after approx 12-24 hours

zzzzzzac, 2021-03-26:
Upgraded to 0.4.0 from 0.3.9, upgrade works but after 12-24 hours (some point in between this) it loses the connection to the synology surevillance station which then requires a reboot of the camera to make it work again. When it has lost connection you can still get to the admin page and its still responding on the network. Rolled back to 0.3.9 now.

kylehase, 2021-03-30:
I believe this is not just related to Synology. 

I'm just using the RTSP stream, without Synology, and noticed that I've been often been unable to load the stream in VLC after upgrading to 0.4.0. I just tried to reboot, per your suggestion, and the stream loaded fine. 

Not sure if it matters but the snapshot continues to work even when RTSP stream fails.

roleoroleo, 2021-04-02:
There are no major changes to the RTSP server in this release.
- Add alaw compression to rRTSPServer.
- Add aac audio to RTSP and improve code.
- Increase h264grabber process priority.
- Decrease audio fifo size to reduce latency.

Please, give me more details:
- Model?
- Audio?
- Resolution?

zzzzzzac, 2021-04-02:
firmware i was using is for h201c, the stream is HD and no audio as dont need this. Its def better on the 0.3.9 as its not lost connection in the last week since reverting back. With 0.4.0 it losing the connection after 12-24 hours for sure. No idea what the difference is between the 2 but i upgraded as it mentioned it had better synology support so was hoping i could use the PTZ in synology but that feature is still not there which is not a big deal as can just use the Yi Home app for that function.
Thanks

roleoroleo, 2021-04-15:
Do you want to test a beta?

About PTZ. you should be able to use it with a Synology. I tested it.

zzzzzzac, 2021-04-15:
sure i can test some beta software, how do you get the ptz working as it does not come up with an option on the DSCAM or surveillance station when configured using rtsp?

roleoroleo, 2021-04-15:
Here it is:
[y203c_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6318002/y203c_0.4.1.tar.gz)

About PTZ. it works only if you configure your cam as ONVIF.

zzzzzzac, 2021-04-15:
i can do onvif if it works but i couldnt get it to work previously, i will download the beta and run it up later.

zzzzzzac, 2021-04-15:
configured onvif but doesnt work on my synology, just trys to activate but then goes disconnected

roleoroleo, 2021-04-15:
What is the problem?
Can you add the cam but ptz doesn't work?
Or you can't add the cam at all?

zzzzzzac, 2021-04-15:
the files in the link are home_y203c and the ones i have are home_h201c is this correct?

the synology test connection worked but then it wont activate and even the test connection does not show any video just a blank screen which is not normal as it usually displays the video once the test is complete.

roleoroleo, 2021-04-15:
No sorry.
[h201c_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6318473/h201c_0.4.1.tar.gz)

Strange behavior, I already fixed this...

zzzzzzac, 2021-04-15:
ok 0.4.1 loaded so will test it out to see if it drops like before after a period of time.

zzzzzzac, 2021-04-15:
the onvif now works also on 0.4.1 :)

zzzzzzac, 2021-04-16:
so this morning its dropped off the synology again like the previous firmware, it works on the yihome app but not on the synology and this time its connected via onvif and not rtsp. It just shows disconnected in the IP camera view on surveillance station, the only way to get it back is to reboot the camera itself.  Disabling and re enabling in synology also does not work.

zzzzzzac, 2021-04-16:
it usually does disconnect quite frequently but always reconnects but at 02:22:11 the camera lost connection but then never reconnected.

zzzzzzac, 2021-04-19:
if you get another version let me know but for now i will have to roll it back to 0.3.9 as it just wont stay connected.

roleoroleo, 2021-04-24:
I moved to the last version of live555 library.
I will send you another beta.

roleoroleo, 2021-04-26:
@zzzzzzac 
Please try this version:
[h201c_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6378309/h201c_0.4.1.tar.gz)

zzzzzzac, 2021-05-02:
just installed, will know by tomorrow if issues of disconnecting is still happening. will post back

zzzzzzac, 2021-05-04:
still disconnecting i am afraid, took a bit longer this time but still the same. The camera shows disconnected in surveillance station but is accessible from the yihome app.

roleoroleo, 2021-05-04:
Okay, let me go step by step.
Save the file rRTSPServer from the version 0.3.9.
Install the last release and try to overwrite the RTSP server only.
I need to identify wich daemon is responsible for the problem.

zzzzzzac, 2021-05-04:
where or how is the rRSTP config exported as if i save config all i get is a zip with these filenames contained within.
TZ
camera.conf
hostname
mqttv4.conf
passwd
proxychains.conf
system.conf

roleoroleo, 2021-05-04:
It's in system.conf

zzzzzzac, 2021-05-04:
so just upload the system.conf through the web page once upgraded to 0.4.1?

roleoroleo, 2021-05-04:
No, you need to save the binary file.
You can find it in /home/yi-hack/bin/rRTSPServer

- Copy it to your pc with an ftp client.
- Update to 0.4.1
- Copy back to the cam your rRTPSServer overwriting the file contained in the 0.4.1

zzzzzzac, 2021-05-04:
ok, now completed. will post in a couple of days to be sure unless it goes before this

zzzzzzac, 2021-05-05:
ok the camera disconnected last night with the rRTSP file from 0.3.9

zzzzzzac, 2021-05-07:
i am now back on 0.3.9 but this one still disconnects but also then reconnects which still makes it more stable than the 0.4.x although of course i would prefer it not to lose connection at all but its better than nothing.
See pic [Here](https://kjconsultantscouk-my.sharepoint.com/:i:/g/personal/zac_zacmorris_co_uk/EfpRNgkVJGVNofrj-zl5D2MBYhBHMHIi_cVra1e3JOMqYg?e=MAtWUo)

roleoroleo, 2021-05-07:
I'm trying to understad which daemon is responsible for the disconnection.
In this moment, which protocol are you using to connect to synology?
rtsp? onvif?

zzzzzzac, 2021-05-07:
up until 2 weeks ago i was using the rstp but now i am using onvif but they disconnect on both protocols but only on synology end. The Yihome app still works when disconnected from synology but the only way to get it back in synology is reboot

roleoroleo, 2021-05-07:
If it's not a problem, please try to use the cam with rtsp, with version 0.4.1 and rRTSPServer from 0.3.9.
If it doesn't work we can exclude rRTSPServer and onvif_srvd.

zzzzzzac, 2021-05-12:
same i am afraid it still disconnects. Its not an issue i will stick with 0.3.9 as this seems to do what I need.

myromeo, 2021-05-13:
Just to chip in, I’m having the same issue, y203c, latest update with ONVIF protocol on my Annke DVR/NVR. Cameras accessible on Wi-Fi and on yi app but show disconnected in NVR. For me a reboot of either NVR or cameras resolved the issue but for extremely short periods of time. Much worse that 3.9 although that still disconnected frequently, just less frequently!

closed 2024-05-09 by github-actions[bot]