roleoroleo/yi-hack-MStar #344: 2 way audio support

NikDevx, 2021-04-07:
Is it possible to add 2 way audio or just audio support for this camera Yi 1080p Home 9FUS (y203c)?

roleoroleo, 2021-04-07:
Audio is supported in the stream.
2 way is work in progress but I don't know if I will be able to complete it.

NikDevx, 2021-04-07:
> Audio is supported in the stream.
> 2 way is work in progress but I don't know if I will be able to complete it.

Can I help you?

I saw something about 2 way audio here https://github.com/Sunoo/homebridge-camera-ffmpeg/issues/738

roleoroleo, 2021-04-08:
The problem here is how to inject audio to the device, not how to stream it.
I'm working on a hacked libalsa but it's very complicated.

NikDevx, 2021-04-08:
Do you mean the problem with detecting the sound card to transfer audio from the device to the camera speaker?

roleoroleo, 2021-04-08:
Audio devices are in use by rmm process.
You can't open them with another process.
About mic, we are using an hacked version of tinyalsa that writes all samples captured to a fifo.
Now, I'm working to a similar hack to play audio on the speaker.

NikDevx, 2021-04-09:
I got it.
Thanks

roleoroleo, 2021-04-15:
Do you want to try a beta?

NikDevx, 2021-04-15:
Yes
I can try tomorrow

roleoroleo, 2021-04-15:
[y203c_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6318002/y203c_0.4.1.tar.gz)

roleoroleo, 2021-04-15:
How it works:
1 - Create a raw pcm (wav without header): 16 KHz, 16 bit, mono (S16LE).
2 - Copy it to /tmp/audio_in_fifo

If you want to use a tts engine download nanotts from here:
https://github.com/roleoroleo/yi-hack-utils
and use with this webservice:

- http://IP-CAM:8080/cgi-bin/speak.sh?lang=en-US
- POST
- Text message in the payload

NikDevx, 2021-04-16:
> How it works:
> 1 - Create a raw pcm (wav without header): 16 KHz, 16 bit, mono (S16LE).
> 2 - Copy it to /tmp/audio_in_fifo
>
> If you want to use a tts engine download nanotts from here: [...]

Which one ver. I should use? MStar, Allwinner, Allwinner-v2

NikDevx, 2021-04-16:
> [y203c_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6318002/y203c_0.4.1.tar.gz)

Will all my configuration data be deleted after flashing?

roleoroleo, 2021-04-16:
Yes, save it in Maintenance page.

NikDevx, 2021-04-23:
Hello!
This working like a charm 
Thanks a lot

NikDevx, 2021-04-24:
Which one audio card you use for send audio to speaker?

For example I can send audio to speaker via ffmpeg
I use `alsa` audio card in my raspberry-pi camera speaker `"audio": "2way -f alsa default"`

**Maybe I can send somehow audio to speaker via ffmpeg?**

roleoroleo, 2021-04-24:
The cam support alsa via tinyalsa lib.
But the device is busy because rmm opens it ad boot.
So, you can't use it with standard programs.
I created a "fork" patching the library: the library reads from a fifo file and send the raw audio to the speaker.

NikDevx, 2021-04-24:
> The cam support alsa via tinyalsa lib.
> But the device is busy because rmm opens it ad boot.
> So, you can't use it with standard programs.
> I created a "fork" patching the library: the library reads from a fifo file and send the raw audio to the speaker.
> You can send audio writing to this file (/tmp/audio_in_fifo) but the stream must to match the correct format: 16 KHz, 16 bit, mono (S16LE).

I already used this one

```
http://IP-CAM:8080/cgi-bin/speak.sh?lang=en-US
POST
Text message in the payload
```

Anyway thank you!

roleoroleo, 2021-04-24:
ffmpeg from local shell or remotely?
Where do you want to run ffmpeg?

NikDevx, 2021-04-25:
ffmpeg on my raspberry-pi
I am using this plugin to add camera to homebridge https://github.com/Sunoo/homebridge-camera-ffmpeg#readme
And this plugin has ffmpeg

roleoroleo, 2021-04-25:
You could try to create a new web service with this code:
```
#!/bin/sh

printf "Content-type: application/json\r\n\r\n"
printf "{}"

cat - > /tmp/audio_in_fifo
```

Pay attention to the permission 0755
Then you can send a file using this web service as a payload of a post request.
I gave it a try and it's ok .
Probably combining ffmpeg and curl it works.

NikDevx, 2021-04-26:
> You could try to create a new web service with this code:
> 
> ```
> #!/bin/sh
> 
> printf "Content-type: application/json\r\n\r\n"
> printf "{}"
> 
> cat - > /tmp/audio_in_fifo
> ```

I should create via ssh or another way?

roleoroleo, 2021-04-26:
https://github.com/roleoroleo/yi-hack-MStar/blob/master/src/www/httpd/cgi-bin/speaker.sh

NikDevx, 2021-04-26:
Thanks a lot!

NikDevx, 2021-04-26:
Can you compile firmware with add speaker service please?

roleoroleo, 2021-04-27:
Check the release section.

NikDevx, 2021-04-27:
> Check the release section.

You are the best!
Thanks!

bepece1, 2021-04-27:
How can I use the new "speaker" function in version 0.4.1???

roleoroleo, 2021-04-27:
**speak.sh**
http://IP-CAM:8080/cgi-bin/speak.sh?lang=en-US
POST
Text message in the payload

**speaker.sh**
http://IP-CAM:8080/cgi-bin/speaker.sh
POST
Raw audio in the payload: 16 KHz, 16 bit, mono (S16LE)

jd1900, 2021-04-27:
Any idea of how to tie this with TTS from Home Assistant to send audio to the camera?

NikDevx, 2021-04-27:
> **speak.sh**
> http://IP-CAM:8080/cgi-bin/speak.sh?lang=en-US
> POST
> Text message in the payload
> 
> **speaker.sh**
> http://IP-CAM:8080/cgi-bin/speaker.sh
> POST
> Raw audio in the payload: 16 KHz, 16 bit, mono (S16LE)

@roleoroleo, on actual version of firmware speak and speaker not working.

my curl :`curl -X POST -d 'hello' http://192.168.0.30:8080/cgi-bin/speak.sh?lang=en-US`
`zsh: no matches found: http://192.168.0.30:8080/cgi-bin/speak.sh?lang=en-US`

work this `curl -X POST -d 'hello' http://192.168.0.30:8080/cgi-bin/speak.sh`
```
{
"error":"false",
"description":"hello"
}
```
But no audio coming

NikDevx, 2021-04-27:
> Any idea of how to tie this with TTS from Home Assistant to send audio to the camera?

This issue not about home assistant
Open new issue and ask

roleoroleo, 2021-04-27:
Install this integration:
https://github.com/roleoroleo/yi-hack_ha_integration/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc
and use speak service.

roleoroleo, 2021-04-27:
> 
> 
> Any idea of how to tie this with TTS from Home Assistant to send audio to the camera?

Install this integration:
https://github.com/roleoroleo/yi-hack_ha_integration/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc
and use speak service.

NikDevx, 2021-04-27:
> > **speak.sh**
> > http://IP-CAM:8080/cgi-bin/speak.sh?lang=en-US
> > POST
> > Text message in the payload
> > **speaker.sh**
> > http://IP-CAM:8080/cgi-bin/speaker.sh
> > POST
> > Raw audio in the payload: 16 KHz, 16 bit, mono (S16LE)
> 
> @roleoroleo, on actual version of firmware service speak and speaker not working.
> 
> my curl: `curl -X POST -d 'hello' http://192.168.0.30:8080/cgi-bin/speak.sh?lang=en-US`
> `zsh: no matches found: http://192.168.0.30:8080/cgi-bin/speak.sh?lang=en-US`
> 
> only work without `lang` parameter `curl -X POST -d 'hello' http://192.168.0.30:8080/cgi-bin/speak.sh`
> 
> ```
> {
> "error":"false",
> "description":"hello"
> }
> ```
> 
> But no audio coming
> 
> ↓↓↓↓↓
> 
> > [y203c_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6318002/y203c_0.4.1.tar.gz)
> 
> Here it was working

@roleoroleo, any updates?

roleoroleo, 2021-04-27:
The only change is this commit: https://github.com/roleoroleo/yi-hack-MStar/commit/3762a570ec91777162228b93d2ac7d4b26e0c1f1
but it's not related.
And with my y203c it works.
I reflashed the 0.4.1 now and tested it with curl.

roleoroleo, 2021-04-27:
Ah...
I moved the path of the tts libs.
/tmp/sd/yi-hack/usr/share...

Before it was:
/tmp/sd/yi-hack/share...

NikDevx, 2021-04-27:
> Ah...
> I moved the path of the tts libs.
> /tmp/sd/yi-hack/usr/share...
> 
> Before it was:
> /tmp/sd/yi-hack/share...

Thanks 🥇

NikDevx, 2021-04-27:
Okay
Service speak working now!

But speaker not play my wav file.

`curl -X POST -F "file=http://192.168.0.30:8080/record/open.wav" http://192.168.0.30:8080/cgi-bin/speaker.sh`
I get in response `{}`
And I hear some crackle

roleoroleo, 2021-04-27:
You need to remove the header from the wav file.
And check if the format is correct.

NikDevx, 2021-04-27:
> You need to remove the header from the wav file.
> And check if the format is correct.

How I can remove header?

Can you give me you wav file for testing speaker, please.

roleoroleo, 2021-04-27:
Remove first 44 bytes.

roleoroleo, 2021-04-27:
You can export a raw data with Audacity.

16 KHz, 16 bit, mono (S16LE)

NikDevx, 2021-04-27:
> Okay
> Service speak working now!
> 
> But speaker not play my wav file.
> 
> `curl -X POST -F "file=http://192.168.0.30:8080/record/open.wav" http://192.168.0.30:8080/cgi-bin/speaker.sh`
> I get in response `{}`
> And I hear some crackle

My open.raw
16 KHz, 16 bit, mono (S16LE)

The same
`curl -X POST -F "file=http://192.168.0.30:8080/record/open.raw" http://192.168.0.30:8080/cgi-bin/speaker.sh`
I get in response {}
And I hear some crackle

The same raw file in `/tmp/audio_in_fifo` work perfect
`curl -X POST -d 'hello' http://192.168.0.30:8080/cgi-bin/speak.sh`
```
{
"error":"false",
"description":"hello"
}
```

NikDevx, 2021-04-27:
I want to make sure I understand correctly
I can send an audio file to the speaker service and it will play my audio file?

foxel, 2021-04-28:
@roleoroleo thanks for the thing. Works great.

The only two things I want to say:
* Please put the instructions for TTS setup somewhere where it's simpler to find. E.g. having your HA integration gives access to TTS immediately even if TTS engine is hot installed
* Is the language set limited by the library? Are there other options for TTS engine which include russian?

foxel, 2021-04-28:
@NikDevx the issue is in your POST payload.
try the following assuming you have `open.raw` file in working directory:
```shell
curl -X POST --data-binary "@open.raw" "http://192.168.0.30:8080/cgi-bin/speaker.sh"
```

foxel, 2021-04-28:
BTW. I was able to use HASS Cloud TTS with YI-hack [here](https://github.com/foxel/yi-hack_ha_integration/tree/hass_cloud_tts).
@roleoroleo this also has some fixes in terms of undefined variables in service calls

roleoroleo, 2021-04-28:
>     * Please put the instructions for TTS setup somewhere where it's simpler to find. E.g. having your HA integration gives access to TTS immediately even if TTS engine is hot installed
> 
>     * Is the language set limited by the library? Are there other options for TTS engine which include russian?

I will add a wiki page.

Languages are limited by the picotts/nanotts library: en, de, es fr and it. But you could find additional packages.
Check this page: https://remzhuk.ru/en/russkii-yazyk-dlya-pico-tts-obzor-sintezatorov-rechi-dlya-android-sintez/

roleoroleo, 2021-04-28:
> 
> 
> BTW. I was able to use HASS Cloud TTS with YI-hack [here](https://github.com/foxel/yi-hack_ha_integration/blob/9003d6c0a2ad3943bd3a782569cfe036e0940347/custom_components/yi_hack/camera.py#L362-L404).
> [image]
> 
> @roleoroleo this also has some fixes in terms of undefined variables in service calls

Thanks for your work, I will check it and add it to my repo, if you agree.

foxel, 2021-04-28:
> Thanks for your work, I will check it and add it to my repo, if you agree.

No problem. You can merge that code. The only reasons stopping me from having a PR right now are the following:
* my solution replaces yours with built-in TTS (maybe it's better to have it optional)
* it requires HASS Cloud subscription
* getting PCM audio with needed parameters is a bit of a hack around hass_nabucasa code which is hardcoded to produce mp3

foxel, 2021-04-28:
Another observation: CGI script playing the audio (`speaker.sh`) streams the body right into the fifo. This has the following effects:
* audio hicckups if data transmission has some lags (encountered only once)
* HTTP request is not closed until the audio is played.

As a fix I use slightly modified `speaker.sh`:
```shell
#!/bin/sh

printf "Content-type: application/json\r\n\r\n"
printf "{}"

cat - > /tmp/speaker_temp

cat /tmp/speaker_temp > /tmp/audio_in_fifo &
```

Not ideal, but works.

roleoroleo, 2021-04-28:
> Another observation: CGI script playing the audio (`speaker.sh`) streams the body right into the fifo. This has the following effects:
>
>     * audio hicckups if data transmission has some lags [...]

The only problem with this solution is the memory.
Probably we should set a limit on the size of the file.
Otherwise the cam may crash and reboot.

roleoroleo, 2021-04-28:
Proposal for a different approach.

We could try to use the standard tts service embedded in ha.
If I'm not wrong the service returns a json like this:
```
{
    "path": "/api/tts_proxy/265944c108cbb00b2a621be5930513e03a0bb2cd_en_-_demo.mp3",
    "url": "http://127.0.0.1:8123/api/tts_proxy/265944c108cbb00b2a621be5930513e03a0bb2cd_en_-_demo.mp3"
}
```
(https://www.home-assistant.io/integrations/tts/)

The component should get mp3 and convert it to the supported format.
Then, send the last file to the web service of the cam.

We should create a new "media_player" platform file with play_media capability.

foxel, 2021-04-28:
Yaeh, that's an option too. Was thinking about it also. I feel something wrong with converting things back and forth, though,

NikDevx, 2021-04-28:
> @NikDevx the issue is in your POST payload.
> try the following assuming you have `open.raw` file in working directory:
> 
> ```shell
> curl -X POST --data-binary "@open.raw" "http://192.168.0.30:8080/cgi-bin/speaker.sh"
> ```

> @NikDevx the issue is in your POST payload.
> try the following assuming you have `open.raw` file in working directory:
> 
> ```shell
> curl -X POST --data-binary "@open.raw" "http://192.168.0.30:8080/cgi-bin/speaker.sh"
> ```

Thank you!
I had an error when trying to open a file from an SD card.
When I open the file from raspberry-pi it works now.

roleoroleo, 2021-04-29:
@foxel 
I'm working on media_player platform.
Do you think it's possible to converto mp3 to raw pcm using ffmpeg (inside ha)?
Or maybe is it possible to install pydub?

roleoroleo, 2021-04-29:
@foxel 
What do you think about this alpha?
The conversion part is very ugly...

```
"""Support for output tts to the yi-hack cam."""
import asyncio
import logging
import subprocess

import requests
from requests.auth import HTTPBasicAuth
from tempfile import NamedTemporaryFile
import os

from homeassistant.components.ffmpeg import CONF_EXTRA_ARGUMENTS, DATA_FFMPEG
from homeassistant.components.media_player import DEVICE_CLASS_SPEAKER, MediaPlayerEntity
from homeassistant.components.media_player.const import (
    SUPPORT_PLAY_MEDIA,
    MEDIA_TYPE_MUSIC,
)
from homeassistant.helpers.device_registry import CONNECTION_NETWORK_MAC
from homeassistant.const import (
    CONF_HOST,
    CONF_MAC,
    CONF_NAME,
    CONF_PASSWORD,
    CONF_PORT,
    CONF_TOKEN,
    CONF_USERNAME,
)

from .const import (
    CONF_SERIAL,
    DEFAULT_BRAND,
    DOMAIN,
)

SUPPORT_YIHACK_MEDIA = (
    SUPPORT_PLAY_MEDIA
)

_LOGGER = logging.getLogger(__name__)

async def async_setup_entry(hass, config_entry, async_add_entities):
    """Set up the Samsung TV from a config entry."""

    async_add_entities([YiHackMediaPlayer(config_entry)])

class YiHackMediaPlayer(MediaPlayerEntity):
    """Define an implementation of a Yi Camera media player."""

    def __init__(self, config):
        """Initialize the device."""
        self._device_name = config.data[CONF_NAME]
        self._name = self._device_name + "_media_player"
        self._unique_id = self._device_name + "_mpca"
        self._mac = config.data[CONF_MAC]
        self._serial_number = config.data[CONF_SERIAL]
        self._is_on = True
        self._host = config.data[CONF_HOST]
        self._port = config.data[CONF_PORT]
        self._user = config.data[CONF_USERNAME]
        self._password = config.data[CONF_PASSWORD]
        # Assume that the media player is not muted
        self._muted = False
        # Assume that the media player is in Play mode
        self._playing = True
        self._state = None

[... 96 lines cut ...]
        open(in_file_path, 'wb').write(r.content)
        _LOGGER.error("Preso file " + in_file_path)

        out_file = NamedTemporaryFile(suffix=".raw", delete=False)
        out_file.close()
        out_file_path = out_file.name
        _LOGGER.error("Creato file 2 " + out_file_path)

        cmd = ["ffmpeg",  "-i",  in_file_path, "-f", "s16le", "-acodec",  "pcm_s16le", "-ar", "16000", "-y", out_file_path]
        subprocess.call(cmd, shell=False)
        _LOGGER.error("Eseguito ffmpeg")

        await self.hass.async_add_executor_job(_perform_speaker, out_file_path)
        _LOGGER.error("Eseguito ws")

        try:
            os.remove(in_file_path)
        except OSError as error:
            _LOGGER.error("Error deleting temp file %s", in_file_path)

        try:
            os.remove(out_file_path)
        except OSError as error:
            _LOGGER.error("Error deleting temp file %s", out_file_path)
```

foxel, 2021-04-29:
@roleoroleo looks like it should work, will test tomorrow.
Btw. Looks like you can do this without temp files with ffmpeg http protocol
https://www.ffmpeg.org/ffmpeg-protocols.html#http

roleoroleo, 2021-04-30:
I created a new "media_player" branch.

foxel, 2021-04-30:
@roleoroleo you forgot to push the code to this new branch )

[49 more replies]

closed 2021-09-24 by roleoroleo