roleoroleo/yi-hack-MStar #345: Hack Works But Can't pair to App

bakersbyteab, 2021-04-10:
Hi my camera updated firmware from vendor - boom all stopped working.
Finally after much searching I found this and the hack worked a treat on my camera. It was down to the chipset and getting that off the bottom of the camera / finding this project.

However I can't pair the camera to yi home app.  It just times out.  
I can control the rotation on the camera from the URL / webpage.  I also imported the feed into ip-cam app.  

Has anyone done anything about controlling the camera  / rotation etc.  I know the PTZ webpage works great.  But from an app at all?  I am just trying to get the camera working similar to how it was under the Yi Home app.

If any ideas or if I need to do anything to the settings to get the pairings to work?  Please share.
THanks
Andy

roleoroleo, 2021-04-10:
Is cloud disabled?
If you disable cloud you can't pair the cam with the app.

bakersbyteab, 2021-04-10:
hi thanks for that I will try the toggle on that as i thought it was off.

bakersbyteab, 2021-04-12:
I tested this with the disable Yi Cloud toggle both grey and blue.  Unfortunately in the add camera although the camera goes onto the wifi and the Yi-Hack web interface works, in Yi Home app it gets stuck at checking pairing success and then times out.  So I have not been able to add the camera.  I can use the web interface but thats not user friendly for camera setting and rotation.  Anyone else seen this at all?  I also tried the polychains-ng toggle but that did not do much either and I don't think is a problem for me..

roleoroleo, 2021-04-24:
But, did it work before to apply the hack?

bakersbyteab, 2021-04-25:
I have a couple of these cameras.  Both worked fine until one had its firmware updated and then stopped working.  Support was useless.  
now I have 1 working camera in the app.  

Hence rather than it being a paper weight I tried this firmware.  It works in the sense I have an ip-camera and the hack made it functional but it still won’t pair to yi home app.   Seeing as the network is not personal against the one device and I have another working have no idea why it won’t pair.

roleoroleo, 2021-04-25:
it's probably better if you try to restore the old firmware.
Which model is it?
y203c, h201c, ...?

bakersbyteab, 2021-04-26:
The model number / chipset is :  BFUSY1EJSK.  so h201c.   However Yi who updated the camera and broke it with the firmware upgrade.

roleoroleo, 2021-04-26:
Use this archive to restore the stock fw:
[h201c.zip](https://github.com/roleoroleo/yi-hack-MStar/files/6374829/h201c.zip)
Let me know when you downloaded the file.

bakersbyteab, 2021-04-26:
Hi thanks for this I have downloaded the ZIP, unpacked the files into root and then flashed the camera again.  This all looked good and I can again get the camera to join wifi OK.  However in the YI home App it won't complete the pairing operation and times out. In the log files produced no obvious errors as to why this final stage of pairing into the app won't work.  Do you have any idea once the wifi is joined what the YI app in pairing status is trying to do?
Thanks

roleoroleo, 2021-04-27:
I don't know.
Try to reset the configuration pressing the button and to start from scratch.

bakersbyteab, 2021-04-28:
Hi,  I have resolved the issue.  My 2nd camera is back working with the application now.  Here is what I did to try and help others.

1. They state the Yi home camera runs on only a 2.4Ghz network.  That is always clear.  What they don't publicise well is that the mobile / app you are trying to pair to must also be on the 2.4Ghz network.
2. For me I am using Google Mesh network.  I had both cameras working fine previously.  Then post firmware changes 1 camera refused to pair.  I would get the wifi connected and then time out on pairing.
3. I could see clearly on all attempts that the camera was connecting to the 2.4Ghz on Google Mesh as it will default to the clients needs - no settings or changes required.
4. My phone however iphone was going onto the 5Ghz channel again by default.

So yesterday I decided to set up a personal hotspot.  I know on older iphones this defaults to 2.4Ghz.  I changed my device name which then sets the wifi hotspot to that of my wifi network,  The password I set to my wifi network also.  This meant I was sure I had a 2.4 Ghz network running.  With my phone not on the network but only on the hotspot I then went through pairing.  This has to be managed to ensure the device does not go to your wifi but the phone hotspot.  Having the phone next to camera helps on that.  You can see the camera connect to the hotspot on the phone as device connected.  Boom the pairing worked and the camera was registered with the right wifi settings plus I could see in the app.   I then disabled the hotspot and reboot the camera (power off/on)  it then joined the network and dadadaaaaa  the camera is still working in the app.

I can only guess that on the other camera and occasions my phone must have dropped to 2.4Ghz on the network to allow the discovery.  The key to note here is BOTH devices need 2.4Ghz and not just the camera.  Older networks you can set this up on specific network but newer routers etc especially google mesh this is automated.

Here is the hotspot set up.  Good for testing also:
1) First step is to make sure you have a device with hotspot compatibilities so we can use it's network temporarily
1) Edit the hotspot network name and password to be exactly the same as Google wifi. - For iphone you go into settings, general and then device name of the phone.  - Temporary change this to the wifi network name.  Also in mobile hotspot before turning on ensure password matches your wifi password.
2) once the hotspot is turned on you will want to turn off google wifi by unplugging. - Not really required if you are willing to have phone next to the device and play about.  However yes ideally remove your wifi from the equation.
3) keep in mind all devices that were connected to google wifi will attempt to jump over to hotspot so I recommend powering off all devices on network except for the one being setup. As temporary bit of overkill but worth noting.
4) Now the hotspot by default only uses 2.4ghz band which the device your setting up can see and connect to.
5) once you have successfully setup device using hotspot network you will then power on the Google wifi and turn off the hotspot. The device will stick to 2.4ghz and will jump back over to google wifi and work as normal.
6) make sure everything is back on and connected and you should be good to go!!

This worked for me.  I now have 2 functioning cameras.
I want to thank all for input and help.  I think the critical element is that for pairing to complete both devices need to be on 2.4Ghz not just the camera!

closed 2021-04-28 by bakersbyteab