roleoroleo/yi-hack-MStar #346: wifi ssid not changed after reboot

divemasterjm, 2021-04-11:
I've tried many times changing ssid from web fronted in many cameras and after reboot always returns to the old ssid.

roleoroleo, 2021-04-24:
I tested again this feature and it works correctly.
Please, give me more details.

divemasterjm, 2021-04-24:
Basically it works but after reboot in a minute turns back to the old ssid 

JuanMa

roleoroleo, 2021-04-24:
Cloud is enabled or disabled?

divemasterjm, 2021-04-24:
disabled

divemasterjm, 2021-04-25:
hi, ive tried with cloud enabled and it works, the problem must be when cloud is disabled
regards

roleoroleo, 2021-04-25:
To understand better due to the double negation.
Cloud disabled = switch enabled?

[image]

divemasterjm, 2021-04-25:
Yes, it is.

closed 2024-05-09 by github-actions[bot]