roleoroleo/yi-hack-MStar #347: IR light / night vision won't turn off

phoenix1887, 2021-04-13:
This seems to be a bigger problem with all the yi cameras as it doesn't work for any of the 12 cameras I have (different models). But I have several with each of the hacks, allwinner, mstar... You can turn the IR light off for any of them in the yi hack settings or via the yi home cloud app but no matter what it doesn't disable night vision, it continues using it. So there seems to be no way to disable night vision on any of the cameras despite the setting. It seems like all of these hacks break the IR / night vision switch.

Can anyone else confirm this IS indeed the case or that the setting is working properly or their cams? By the way cycling the lights on and off and rebooting the camera doesn't help...

To test it out disable the IR light light setting on the camera then dim the lights in the room until you see the camera switch to night vision... Even though it shouldn't...

Anyone have any ideas how to fix this?

samtap, 2021-05-07:
I'd also like to get full control over night vision mode.

Usually night-vision involves three steps:
- Power on IR leds
- Remove the IR filter from lens, this causes the typical 'click' sound
- Disable saturation, this causes the black and white image

The first step is easy, probably just setting a gpio pin and already available via a toggle switch on the web-interface. I've been trying to find a method to do the other two manually via command-line tool or sysfs. Also the light-sensor needs to be disabled or it will just revert back depending on ambient light levels. I don't think generic sysfs entries are common for MStar unfortunately, so a custom tool may be required, examples should be in MStar SDK.

roleoroleo, 2021-05-07:
AFAIK ir filter is toggled by gpio 21 and 22.
I didn't find ir led.

chitown60608, 2021-07-23:
Would LOVE this feature.  I don't care if it's an option to always have the IR on or always off, or the ability to turn it on/off.  I use this as an infant cam and from about 5:50 in the morning until 7:45, this thing cycles on and off constantly.  It's so loud that I have to turn it off during that time or it wakes the baby.  Please, someone with infinitely more skills than myself, fix this.

chitown60608, 2021-07-24:
Also, guess I should've specified;  I need the IR cut filter disabled.  Just turning off the IR LEDs won't do anything.  It's the cut filter that makes the clicking noise.  The first gen Yi camera had the option to disable it.  All newer cameras only allow the IR lights to be turned off.

chitown60608, 2021-09-14:
Thought I would add a couple of screen shots to illustrate the option that was there for the older YI cameras.  Looks like it was there with FW 2.1.1_20171024151200 and FW 2.1.1_20160301202600 and probably all in between those dates.  Thought maybe someone could look at the code for those two and possibly implement it for the newer cams.

[image]

[image]

SpinyUK, 2021-10-04:
I would also like to see an option to disable the drop in and out of  'night mode' - I have a camera in my bedroom that looks out to my car, and the clicking as it cycles when the sun comes up is annoying

janekjelonek, 2021-11-18:
If anyone ever finds the way to quieten this clicking - **_please let us know_**.  
By any means necessary. 
Regardless of whether I should use either software hack, a solder, a drill or ducktape. 

Cheers, 
Janek

mortenmoulder, 2021-12-15:
I'm experiencing the same thing. Even when I turn on the lights in the room, it can take a few minutes before it turns satuation on again. Quite annoying to say the least, because the quality is already quite poor, so having lights + no satuation is really bad.

nelshi, 2021-12-18:
Hi,

I thought I would reply as this is what came up when I was searching for a resolution to the annoying "clicking" sound when my Yi Home 1080p would switch to Night Mode/Greyscale.
Mine is located within a bedroom so the clicking sound was a nuisance early in the morning when the camera flicked in and out of night mode rapidly as the light level changed.
I ended up opening the camera itself and seem to have identified the cable which enables the clicking to occur - circled in the image below.
Upon disconnecting that cable and reassembling my camera I no longer have the "clicking" occurring when it switches to greyscale mode, or upon boot of the camera.
Hopefully this ends up helping someone - this may break something else, but for my usage the camera seems to be working as normal so far.

[image]

SpinyUK, 2021-12-18:
> Hi,
>
> I thought I would reply as this is what came up when I was searching for a resolution to the annoying "clicking" sound when my Yi Home 1080p would switch to Night Mode/Greyscale. Mine is [...]

nice find :) I suspect the 'clicking' is the IR filter moving in and out of the front of the lens so it fiters ir in 'day' mode and doesn't in 'night mode.

but to confirm my take on the request, i'd like an option to force the camera to stay in 'day' mode - I have a Reolink RTSP camera (can't remember the model, sorry!) that has this option, and it's super useful.

five2seven, 2022-09-07:
I've also been searching for a way to make this clicking stop!

lenisko, 2024-01-12:
First get filter to desired position. Plug off the microUSB. If there was no click, just disconnect connector. Should stay at this position.

closed 2024-05-13 by github-actions[bot]