roleoroleo/yi-hack-MStar #348: Bricked on update (9FUS)

Schany79, 2021-04-15:
Hello,

my 9FUS bricked after update from 0.2.2 to 0.2.5. It doesn´t start anymore.
I´ve tried to flash it again with all kinds of possible files for 9FUS.
The only thing what the camera does is blinking in yellow.

Can you please help me here?

Thanks,
Sascha

roleoroleo, 2021-04-15:
?!?
Is it a MStar cam?

Schany79, 2021-04-15:
It´s a Yi-Home-1080 with Serial 9FUS... 
Where can I see if it is a MStar?

roleoroleo, 2021-04-15:
It depends on the chip used but you can see reading original fw version.
If you are using hack 0.2.2 probably it's not a Mstar.

Schany79, 2021-04-15:
I think from your Git. For a year or so...
The Cam has the 0.2.2 Firmware and under Maintenance (on the Cam´s Website) I tried to update the cam to 0.2.5
I think my SD Card crashed while the update started. Then I tried it with your newest Version for the 9FUS.

Schany79, 2021-04-15:
Firmware (Original) is 8.20.0A_202007211802 (I have 4 Cams with the exact same firmware - all the same cams)

roleoroleo, 2021-04-15:
Your cam requires this hack:
https://github.com/roleoroleo/yi-hack-Allwinner

Schany79, 2021-04-15:
Oh Man! That´s right! Thank you.... It works!

closed 2021-04-15 by Schany79