roleoroleo/yi-hack-MStar #350: Static IP

hetzbh, 2021-04-19:
Hi,

Is there a way to set the IP to static and set the parameters? it's a bit frustrating to hunt the IP of each camera...

Thanks

roleoroleo, 2021-04-20:
https://github.com/roleoroleo/yi-hack-MStar/issues/208

hetzbh, 2021-04-20:
While the #208 is a good manual solution, it looks like the app itself does not update the IP, only the parts that you added (web GUI etc) are updated.

roleoroleo, 2021-04-20:
I don't understand, sorry.

SeriousPat, 2022-02-06:
> #208

Does not work: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/347

SeriousPat, 2022-02-11:
I found out, that the IP Bug i reported has to with my DLAN AP.
Without the DLAN AP, the Cams get everytime the right reserved ip from my router.
but i can test it for you, if you want that

roleoroleo, 2022-02-11:
Maybe there are 2 dhcp servers on your lan.

> but i can test it for you, if you want that

Yes please, if you have time.

closed 2024-05-09 by github-actions[bot]