roleoroleo/yi-hack-MStar #351: HA integration

roleoroleo, 2021-04-21:
Please, take a look at this project:

https://github.com/roleoroleo/yi-hack_ha_integration

foxel, 2021-04-21:
Cool :+1: 

* https://github.com/roleoroleo/yi-hack_ha_integration/blob/9d370701f728357a40ca710bbab1e796ea2e2f39/yi-hack/camera.py#L61 - suffix looks strange, shouldn't it start with an underscore?
* https://github.com/roleoroleo/yi-hack_ha_integration/blob/9d370701f728357a40ca710bbab1e796ea2e2f39/yi-hack/camera.py#L72 - should be `self._port`
* https://github.com/roleoroleo/yi-hack_ha_integration/blob/9d370701f728357a40ca710bbab1e796ea2e2f39/yi-hack/binary_sensor.py#L178-L187 - duplicate code
* https://github.com/roleoroleo/yi-hack_ha_integration/blob/9d370701f728357a40ca710bbab1e796ea2e2f39/yi-hack/config_flow.py#L57-L68 - all of these variables are local to `async_step_user` and not needed to be members of the flow instance

Since topics are configurable after the integration is configured it looks like you need to read topics at least inside `async_setup_entry` so that users don't need to remove and re-add the integration after such changes.

roleoroleo, 2021-04-21:
Thak you for your notes.
I'm taking my first steps in ha programming and I need guidance.

roleoroleo, 2021-04-21:
Done.
About the last step, I need to study it.

roleoroleo, 2021-04-23:
I fixed the last step but I have a weird problem after this change.
When the wizard completes, ha doesn't ask the destination area for the new entities.
I have to set it later.

divemasterjm, 2021-04-25:
i ha to comment AI HUMAN and SOUND in init_py cause integration throw an error with this entities. after comment worked ok

roleoroleo, 2021-04-25:
What's your model?
mstar, allwinner or allwinner-v2?

divemasterjm, 2021-04-25:
mstar

roleoroleo, 2021-04-25:
Ok, fixed AI HUMAN and SOUND.
Thanks.

roleoroleo, 2021-04-25:
What do you mean with "i miss binary_sensor to activate/desactivate motion events"?

divemasterjm, 2021-04-25:
as per your integration you can have only few sensor, will be nice to have as most as you can just for setting up automations, i mean for binary_sensor for activation of motion events for having a sensor to activate or deactivate motion enevts just to for example setup an automation to record and have snapshots on/off

foxel, 2021-04-26:
@divemasterjm the thing you are asking for should be performed with one binary sensor. In HA you can trigger automation on both events where it goes from no motion to motion detected and vice-versa

divemasterjm, 2021-04-26:
@foxel i know but i want to deactivate motion detection based on state if for example alarm is amerd home, night, away, therefore i only have snapshot or movie when this setting is on

roleoroleo, 2021-04-26:
Sorry but I don't understand.
Could you explain me better?

divemasterjm, 2021-04-26:
i want to have ftp uploads of  images or videos que i turn switch ftp updload ON
this why i ask for having more switches in the integration to control camera

FTP Upload | Enable/disable upload of recorded events to a FTP service.
-- | --

closed 2024-05-09 by github-actions[bot]