roleoroleo/yi-hack-MStar #357: Video not recording after upgrading to 0.4.1

jd1900, 2021-05-05:
Last 1st of May I updated to 0.4.1.

Since then, I receive the motion events, but there are no new recording. The last recording is from the moment before the update.

I didn't change anything in the configuration.

I've tried disabling and enabling recording with no luck.

Any idea of how to fix this?

jd1900, 2021-05-05:
I've just realized that I can't access the official app.

roleoroleo, 2021-05-05:
If you want to use the app, uncheck the option "Disable cloud".

jd1900, 2021-05-05:
It is unchecked.

[image]

roleoroleo, 2021-05-07:
Did you solve the problem?

jd1900, 2021-05-07:
No. How could I?

roleoroleo, 2021-05-07:
The cam is configured as in the image above and you can't connect with yi app.
Correct?
Does rtsp work?

jd1900, 2021-05-07:
Correct.

I can see the image of the video and screenshots, but I can't see new recording or use the app.

roleoroleo, 2021-05-08:
I have never seen a problem like this.
Try to downgrade to 0.4.0 and check if it works.

jd1900, 2021-05-08:
Is there a way to do that or reinstall 0.4.1 without having to extract the SD card?

It isn't quite accesible after installing in the cealing.

jd1900, 2021-05-08:
One weird thing.
I've tried disabling cloud, and the time is wrong:
[image]

I have NTP enabled, but it doesn't work.

However, when I re-enable the cloud back, it works again:
[image]

I've tried when disabling the cloud every possible configuration, but still no luck.

jd1900, 2021-05-08:
One more thing. It stuck in 25% when trying to connect in the official app. When the cloud is disabled it stucks in 5%

roleoroleo, 2021-05-08:
> 
> 
> Is there a way to do that or reinstall 0.4.1 without having to extract the SD card?
> 
> It isn't quite accesible after installing in the cealing.

Copy the files of the old version in /tmp/sd through ftp.
And restart.

roleoroleo, 2021-05-08:
Are you sure that the cam is able to connect to internet?
Wifi signal, firewall, etc...

jd1900, 2021-05-08:
> Are you sure that the cam is able to connect to internet?
> Wifi signal, firewall, etc...

It seems that the problem is **related to the DNS server**.
I can't reach any server at all:
[image]

However, the DNS server is well set in /tmp/dns1 and /tmp/resolv.conf

Furthermore, if I created a hosts file and add servers, the manually added servers work.

jd1900, 2021-05-08:
After downgrading everything works back again, but I've lost all my config.
I luckily backuped it up. 
I have upgraded back again, and restored the configuration.

It seems to work fine back again, at least the DNS.

Thank you!

closed 2021-09-24 by roleoroleo