roleoroleo/yi-hack-MStar #358: Couple of question re YI Dome Camera 1080p

MadTomT, 2021-05-06:
Hi. Is this model supported ?
Serial starts BH201CE......
DID starts: BFUSY1EIG...

Doing a pcap on the network now shows the camera is sending data to 170.106.6.47, if I apply this firmware will that stop ?
Do the move features work ?

Thanks

roleoroleo, 2021-05-06:
What's the fw version?

MadTomT, 2021-05-06:
Hi. Firmware version 4.6.0.0A_201908271549

Thanks

roleoroleo, 2021-05-06:
https://github.com/roleoroleo/yi-hack-MStar

MadTomT, 2021-05-06:
Thanks. Do you know which firmware I'd need to use ?

roleoroleo, 2021-05-07:
h201c

MadTomT, 2021-05-07:
Cheers. I'll try this later today 🙂
Does this firmware require internet access to run, or can I restrict the camera to the WiFi network only ?
Is there an option to go back to the original firmware if needed.?  Thanks

roleoroleo, 2021-05-07:
> Does this firmware require internet access to run, or can I restrict the camera to the WiFi network only ?
Yes, if you don't want to use yi app.

> Is there an option to go back to the original firmware if needed.? Thanks
Yes, if you have a backup of the original fw.
See the wiki.

MadTomT, 2021-05-07:
Thanks. I've had a look at the wiki and I don't really want to start soldering cables to the camera.
Do you know if any copies of the firmware are available to roll back if needed ?

Thanks

roleoroleo, 2021-05-07:
Maybe, check the issues.

MadTomT, 2021-05-08:
Decided to flash it anyway.. All installed and seems to be working well with Onifver and TinyCam on Android.
Do you know if its possible to get 2 way audio ? I know TinyCam Pro supports it, but does the cam / firmware ?

Thanks

roleoroleo, 2021-05-08:
2 way audio is available only through the speaker web service.
I don't know what protocol Tinycam uses.

MadTomT, 2021-05-09:
Hi. I'm trying to find out what tinycam uses.. would be interesting to know.

How do I access it via the web service ? I can't see a link for it.
Thanks

roleoroleo, 2021-05-09:
See the wiki about web services.