roleoroleo/yi-hack-MStar #359: Fix timezone for Moscow city

Smosia, 2021-05-09:
Looks like timezone for Moscow city (MSK-4) is wrong. 
In Moscow there is UTC+3 timezone, not UTC+4

roleoroleo, 2021-05-09:
Ok, I will fix it.

Smosia, 2021-05-09:
hm.. Strange behaviour. 
Camera time become utc-3 when I wrote UTC+3 in web gui. (UTC = 10:00, Camera time = 7:00)
Then I change it to UTC-3 and everything works (UTC = 10:00, Camera time = 13:00)

Looks like there are some problems with number sign.

foxel, 2021-05-09:
Please read this https://unix.stackexchange.com/questions/104088/why-does-tz-utc-8-produce-dates-that-are-utc8

Smosia, 2021-05-09:
@foxel Thank you for explanation.

In https://github.com/roleoroleo/yi-hack-MStar/commit/66421d4b85f07d65e7d3092639bcce5f9e678910 should be UTC-3 timezone.

roleoroleo, 2021-05-09:
Ok.

foxel, 2021-05-09:
As I understand the abbreviation is the desired name of time zone. In our case should be 'MSK-3'

Smosia, 2021-05-09:
@foxel You're right. MSK-3 works too.

closed 2021-09-24 by roleoroleo