roleoroleo/yi-hack-MStar #360: Toggle "Switch on/off the camera" in web GUI is not working

Smosia, 2021-05-09:
Looks like toggle "Switch on/off the camera" in web GUI is not working.
When I turn off the camera with that toggle rtsp steam and motion detection process continues.

My camera is: Yi home camera 3 9FUS. 
Software version: 4.2.0

roleoroleo, 2021-05-09:
I know, the maning of this switch is not clear.
It doesn't switch on/off the stream but it replicates the feature of the switch in the original yi app.

Smosia, 2021-05-09:
Looks like there is no way to disable video from camera without rebooting. 

Any proposals about it?

Smosia, 2021-05-09:
Is it possible to add new command to mqtt advertise that would call start_rtsp(), stop_rtsp() from service.sh?
Or there is another way to call these functions from HomeAssistant?

roleoroleo, 2021-05-10:
https://www.home-assistant.io/integrations/rest_command/

Rushmed, 2021-05-12:
> I know, the maning of this switch is not clear.
> It doesn't switch on/off the stream but it replicates the feature of the switch in the original yi app.

For me its still not clear what this switch is doing. Could you please explain?

roleoroleo, 2021-05-12:
It enable disable the communication between the app and the cam.
Stream, audio and detections.

closed 2021-09-24 by roleoroleo