roleoroleo/yi-hack-MStar #364: Crontab without asterisks?

tulkas84, 2021-05-27:
Hi, 
I am triying to set up a cron for reboot every day but the web interface only allows this special chars: \ | ! " $ % & / ( ) ' ? ^ [ ] @ < > , ; . : - _ 

Is a mistake or i am missing something? I want to write "0 1 * * * /sbin/reboot"

Thanks!

roleoroleo, 2021-05-27:
You are right :)
The asterisk is allowed in the crontab field.
But there is a bug at the moment:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/77#issuecomment-848963138

roleoroleo, 2021-05-28:
Maybe I fixed it with this commit:
https://github.com/roleoroleo/yi-hack-MStar/commit/c5ba66ba89983ae7a5dab40fd65788772448552e

galinowski, 2021-06-01:
I manually tested your fix c5ba66b and it worked, thanks!

closed 2021-09-24 by roleoroleo