roleoroleo/yi-hack-MStar #365: RTSP stream stuck

radokristof, 2021-05-28:
I discovered that after some time (few mins) the RTSP stream is inaccessible in my camera.

VLC just spins indefinitely and can't show the stream. If I restart the camera it works for a little time.
ffmpeg shows:
```
rtsp://10.0.1.11/ch0_1.h264: Invalid data found when processing input
```
ffmpeg command:
```
ffmpeg -re -rtsp_transport tcp -err_detect ignore_err -i rtsp://10.0.1.11/ch0_1.h264 -acodec aac -vcodec copy test.mp4
```
I have tried ```ffmpeg``` with different parameters it didn't worked. ```ffmpeg``` stuck right after the headers are printed to the console, so I couldn't get more info on this even with ```-d 7```

If I restart the camera ffmpeg also works correctly.
Maybe the camera doesn't have enough resources? Because I also use the FTP client feature that uploads the recorded files to an FTP server.

Thanks in advance!

roleoroleo, 2021-05-28:
Has the problem appeared recently?
Maybe after an upgrade?

radokristof, 2021-05-28:
Well, I can't say that for sure.
I have used the RTSP stream to show the video feed in my home automation system (which basically uses ffmpeg), but it stopped working sometime.
I did not had time to look into the issue and I suspected that something was wrong on the client side (I mean the server which fetches the RTSP stream - because I had problems with it before). I had some little time to test it now and it seems broken.
I'm on 0.4.1 now, I will try to downgrade to 0.4.0, this is what I used to have.

radokristof, 2021-05-28:
How can I downgrade? Is it safe to downgrade further? Like 0.3.9?

roleoroleo, 2021-05-28:
Yes 0.3.9 and yes it's safe.
But it's better to reconfigure the cam from scratch.

closed 2022-04-08 by radokristof