roleoroleo/yi-hack-MStar #366: Sending audio to Camera

MadTomT, 2021-05-29:
Hi.
Can you explain how to send an audio file or speech to the camera.
I've had a look at the wiki, but I'm not sure what I need to do..

Thanks

roleoroleo, 2021-05-29:
Audio to the speaker is sent through 2 web services.

1 - Use the internal TTS engine to play a sentence (if TTS is installed)
http://IP:8080/cgi-bin/speak.sh?lang=LANG
where LANG=language (en-US, en-GB, de-DE, es-ES, fr-FR, it-IT)

You have to send a http post message and add the sentence you want to speech in the payload.

2 - Play an audio file
http://IP:8080/cgi-bin/speaker.sh
You have to send a post message and add the raw audio file in the payload. The format must be 16 KHz, 16 bit, mono (S16LE).

roleoroleo, 2021-05-29:
If you use Home Assistant, you can use standard tts engine with this integration.
https://github.com/roleoroleo/yi-hack_ha_integration

MadTomT, 2021-05-29:
Thanks. is it possible to provide an example of how to send the raw audio to speaker.sh ?

roleoroleo, 2021-05-30:
Check line 145:
https://github.com/roleoroleo/yi-hack_ha_integration/blob/main/custom_components/yi_hack/media_player.py

MadTomT, 2021-06-04:
Thanks for the link, still not sure what to do with this.

roleoroleo, 2021-06-06:
What tool do you think to use for sending audio to the cam?

MadTomT, 2021-06-07:
HI. I was hoping to try and select a file in a web page and submit to the camera.
Alternatively curl at the command line would work 

Thanks

roleoroleo, 2021-06-07:
curl -i -X POST http://192.168.1.200:8080/cgi-bin/speakers.sh --data-binary "@audio_file.raw

MadTomT, 2021-08-02:
Sorry for the delay in trying this.
When I run the curl command containing my IP and raw file I get :

HTTP/1.0 401 Unauthorized
Date: Fri, 17 Apr 2020 20:18:39 GMT
Connection: close
Content-type: text/html
WWW-Authenticate: Basic realm="Web Server Authentication"

<HTML><HEAD><TITLE>401 Unauthorized</TITLE></HEAD>
<BODY><H1>401 Unauthorized</H1>
</BODY></HTML>

How do I authenticate via curl ?

roleoroleo, 2021-08-04:
If you enable http auth you need to add your credentials.
Check -u option.

tulkas84, 2021-08-31:
Hi, I get this error when using curl from SSH: 

```
$ curl -i -X POST http://192.168.0.56:8080/cgi-bin/speak.sh?lang=es-ES -d "test"
HTTP/1.0 200 OK
Content-type: application/json
{
"error":"true"
"description":"TTS engine not found"
}
```
I dont find any doc about enablig TTS. Any clue? Thanks :)

roleoroleo, 2021-09-01:
Install the tts engine from here:
https://github.com/roleoroleo/yi-hack-utils/

closed 2021-09-24 by roleoroleo