roleoroleo/yi-hack-MStar #368: Help on hack version - model: YYS.H718

vin-w, 2021-06-01:
I have a xiaoyi camera chinese model, it has a name called xiaoyi H7, the look is like yi dome X.
https://www.xiaoyi.com/homeCamera/homeCamera_h7/index.html

[image]

Model Name: YYS.H718
Serial Number: 6FCNYXXXXX

When i looked into the log_wifi_connected.tar.gz

```
homever
---------
4.4.0.0A_201809101611

df.txt
---------
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/block/mtdblock3     12480      7860      4620  63% /home
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 1920      1764       156  92% /

file.txt
---------
/home:
-rw-rw-r--    1 1005     1005            22 Sep 10  2018 homever
drwxr-xr-x    7 root     0                0 Jan  1  1970 .
drwxr-xr-x   13 root     0                0 Jan  1  1970 ..
drwxrwxr-x    9 1005     1005             0 Jan  1  1970 app
drwxrwxr-x    5 1005     1005             0 Sep 10  2018 base
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 lib
drwxrwxr-x    3 1005     1005             0 Sep 10  2018 ms

/home/app:
-rwxrwxr-x    1 1005     1005       1047240 Sep 10  2018 oss
-rwxrwxr-x    1 1005     1005        565396 Sep 10  2018 rmm
-rwxrwxr-x    1 1005     1005        376500 Sep 10  2018 cloudAPI
-rwxrwxr-x    1 1005     1005        310248 Sep 10  2018 p2p_tnp
-rwxrwxr-x    1 1005     1005         81704 Sep 10  2018 cloud
-rwxrwxr-x    1 1005     1005         51308 Sep 10  2018 dispatch
-rwxrwxr-x    1 1005     1005         37160 Sep 10  2018 mp4record
-rwxrwxr-x    1 1005     1005         27704 Sep 10  2018 yi1080.bmp
-rwxrwxr-x    1 1005     1005         26724 Sep 10  2018 crypt_file
-rwxrwxr-x    1 1005     1005         20783 Sep 10  2018 watch_process
-rwxrwxr-x    1 1005     1005         14720 Sep 10  2018 log_server
-rwxrwxr-x    1 1005     1005         13228 Sep 10  2018 upgrade_firmware
-rwxrwxr-x    1 1005     1005         12680 Sep 10  2018 arp_test
-rwxrwxrwx    1 1005     1005          4799 Sep 10  2018 init.sh
-rwxrwxr-x    1 1005     1005          3128 Sep 10  2018 yi360.bmp
-rwxrwxr-x    1 1005     1005           441 Sep 10  2018 wp_cmd
-rw-rw-r--    1 1005     1005            22 Sep 10  2018 .appver
lrwxrwxrwx    1 root     0                3 Jan  1  1970 oss_fast -> oss
lrwxrwxrwx    1 root     0                3 Jan  1  1970 oss_lapse -> oss
drwxrwxr-x    9 1005     1005             0 Jan  1  1970 .
drwxr-xr-x    7 root     0                0 Jan  1  1970 ..
drwxrwxr-x    6 1005     1005             0 Sep 10  2018 audio_file
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 ca
drwxrwxr-x    5 1005     1005             0 Sep 10  2018 data
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 localbin
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 localko
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 locallib
drwxrwxr-x    2 1005     1005             0 Jan 19 06:42 script

/home/app/audio_file:
drwxrwxr-x    6 1005     1005             0 Sep 10  2018 .
drwxrwxr-x    9 1005     1005             0 Jan  1  1970 ..
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 common
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 jp
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 simplecn
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 us
[... 239 lines cut ...]
-rwxrwxr-x    1 1005     1005          3608 Sep 10  2018 libyuvmon.so
-rw-rw-r--    1 1005     1005            22 Sep 10  2018 .msver
drwxrwxr-x    3 1005     1005             0 Sep 10  2018 .
drwxr-xr-x    7 root     0                0 Jan  1  1970 ..
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 ko

/home/ms/ko:
-rwxrwxr-x    1 1005     1005         77784 Sep 10  2018 mdrv_mvhe.ko
-rwxrwxr-x    1 1005     1005         48908 Sep 10  2018 mdrv_mmfe.ko
-rwxrwxr-x    1 1005     1005         26200 Sep 10  2018 mstar_jpe.ko
drwxrwxr-x    2 1005     1005             0 Sep 10  2018 .
drwxrwxr-x    3 1005     1005             0 Sep 10  2018 ..

```

I tried to extract y30_0.4.1.tgz content into root of SD card
but both home_y30 + sys_y30 did not rename to .done

Any idea which version should I use?

I guess it should be mstar because 
/home/app/data/audio/mstar_mic.cfg
/home/ms/ko/mstar_jpe.ko

Many thanks!

roleoroleo, 2021-06-02:
If the fw is 4.4.0, this cam is not currently supported.
If you want to help me to support it I need a dump of the flash.
Read the wiki.

vin-w, 2021-06-02:
@roleoroleo , thanks for your reply!

So I must open it and dump via cable?

Possible to dump with backup.tar.gz? Does this method support on mstar?

roleoroleo, 2021-06-02:
AFAIK this method doesn't work in mstar platform.
Sorry.

vin-w, 2021-06-03:
@roleoroleo, thanks for confirming. I tried, but the camera keep flashing yellow with no backup directory created too.

I will try to open the Yi Dome X and dump the flash.

vin-w, 2021-06-03:
Repost in here:

I am trying to open my Yi Dome X
The screw at the back of the camera under the speaker is removed.

[image]<br />

[image]

Anyone knows how to remove the white cover and make the PCB visible?

vin-w, 2021-06-14:
@roleoroleo 
I was able to connect to UART.
But I failed the dump command from your instruction in wiki
https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware

Look forward to your help on this.

Here is my boot log:
https://pastebin.com/yd1qZJXz

For the address it is the same mentioned in the wiki.
```
[22:27:33]Creating 6 MTD partitions on "NOR_FLASH":
[22:27:33]0x000000000000-0x000000050000 : "BOOT"
[22:27:33]0x000000050000-0x0000001d0000 : "KERNEL"
[22:27:33]0x0000001d0000-0x0000003b0000 : "ROOTFS"
[22:27:33]0x0000003b0000-0x000000fe0000 : "HOME"
[22:27:33]0x000000fe0000-0x000000ff0000 : "vd1"
[22:27:33]0x000000ff0000-0x000001000000 : "conf"
```
So I tried to dump sys_xxx.

With 
1. My camera did not connect to the yi home app before (i.e. wifi not connected)
2. SD card was not plugged

Now
at step3:
Stop the boot at the U-Boot prompt (should be "MStar #") pressing "Enter".

I never able to see "MStar #" in my boot log.
I tried to press a lot of enter many times during boot up, but it just flush the log until the end (see the pastebin)

So I continue to type the command
```
sf probe <ENTER, nth shown on screen>
sf read 0x21000000 0x1d0000 0x1e0000 <ENTER, nth shown on screen>
```

Quit terminal

```
$ python3 dump.py --timeout 1 --step 64 /dev/cu.usbserial-142120 115200 0x21000000 0x211dfff0 --debug
Debug mode enabled
Debug write: md 0x21000000 0x100

Debug read: []
Traceback (most recent call last):
  File "dump.py", line 266, in <module>
    main()
  File "dump.py", line 250, in main
    raise e
  File "dump.py", line 248, in main
    data = dump(serial, opts, log)
  File "dump.py", line 136, in dump
    assert len(chunk) > 0
AssertionError
```

roleoroleo, 2021-06-14:
Reading your log, you didn't enter into the bootloader.
You should enter into the bootloader before the line:
`[22:27:32]Starting kernel ...`
If the cam doesn't stop you can't continue with the procedure.

Please try to press Enter immediately. If doesn't work, check if the both pins (rx/tx) are properly connected (maybe you are able to receive but not to send).

vin-w, 2021-06-14:
@roleoroleo Thanks again. Finally able to enter the command prompt

```
MStar # sf probe
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
MStar # 
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
MStar # sf read 0x21000000 0x3b0000 0xc30000
SF: 12779520 bytes @ 0x3b0000 Read: OK
MStar # 
SF: 12779520 bytes @ 0x3b0000 Read: OK
```

and dumping in progress now!

vin-w, 2021-06-15:
@roleoroleo I have dumped successfully

Here it is:
https://www.dropbox.com/s/154odpbiczdpopa/dump.zip?dl=0

Thank you!

roleoroleo, 2021-06-16:
It seems there is a problem with your sys_h307.img.swapped file.
It doesn't contain sys partition but a piece of the home partition.
Maybe a problem during the dump?
Maybe you changed the size but not the offset in the command
`sf read 0x21000000 0x1d0000 0x1e0000`

vin-w, 2021-06-17:
@roleoroleo Here is the dump again.
https://www.dropbox.com/s/jase8xxo0uenegj/dump_again.zip?dl=0

roleoroleo, 2021-06-17:
Do you want to try this hack?
[h307_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6670779/h307_0.4.1.tar.gz)

vin-w, 2021-06-17:
@roleoroleo Confirmed it's working! Thanks a lot!
But it is missing the switch of "Swap File" in the web page.

Other thing, in this camera, there is an option to set the video codec: H264/H265.
Everything is ok so far when I used H264.

Will test later with H265.

roleoroleo, 2021-06-17:
I think swap file is not necessary in mstar platform?
Did you notice memory problems or unexpected reboots?

H265 will not work without changes.

roleoroleo, 2021-06-17:
Please try if ptz works.

vin-w, 2021-06-21:
@roleoroleo, thanks for the check. Here is my test report.

With h264 setting in app
- high/low res snapshot
PASS (most of the time)
but a few times calling snapshot is quite slow and it crashed the cam.

- rtsp high/audio snapshot
FAILED (always reproducible)
rtsp high -> audio disappear after a few sec
rtsp audio only -> audio disappear after a few sec

my setting:
RTSP Stream | High
RTSP Audio | alaw

- ptz 
PASS

- Other thing to note:
I ran it over the weekend and found that it power off itself after running for 1 day. (same issue for 2 days already)

roleoroleo, 2021-06-22:
>     * high/low res snapshot
>       PASS (most of the time)
>       but 1-2 times calling snapshot (web url) is quite slow and it crashed the cam.
> 

How do you run the snapshot? From the web interface?

>     * rtsp high/audio snapshot
>       FAILED (always reproducible)
>       rtsp high -> audio disappear after a few sec
>       rtsp audio only -> audio disappear after a few sec

Which client are you using?

>     * tts to speed from hass to yi cam
>       FAILED, crashed a few times (1~3 sec speech)

Do you mean yi_hack.speak or tts (google/alexa) service?

vin-w, 2021-06-22:
> How do you run the snapshot? From the web interface?

http://<IP>:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
http://<IP>:8080/cgi-bin/snapshot.sh?res=low&watermark=yes

when i keep taking snapshot in around 5 sec per take, it sometimes could crash.

> Which client are you using?

**RTSP high (setting: ulaw audio)**
VLC: stop audio after a few sec (the media info said it is mlaw, incorrect!)
HASS: webrtc integration: voice is distorted
yi home app: clear voice

**RTSP high (setting: alaw audio)**
VLC: clear voice (the media info said it is alaw, correct)
HASS: webrtc integration: voice is distorted
yi home app: clear voice

> Do you mean yi_hack.speak or tts (google/alexa) service?

tts service on hass, it is quite easy to crash
combine with the problem of snapshot web link, I wonder if there is some problem/load problem in the web server receiving the request?

One more thing: I think the serial number was not shown correctly?
"serial_number": Starting with >> "XsaOQ4 (should be same value as bottom of camera QR?)

vin-w, 2021-06-22:
I just accidentally clicked "Reset to yi-hack default" 
and then my camera cannot connect to app.

I tried to remove the camera from app and then restore to dump firmware.
The cam did flash orange led > then it say this camera cannot be used in china
From the SD card, it used my previously saved wifi credential
It didn't ask me to scan for barcode
How could I reset and let the app able to scan wifi QR scan again?

I now restored to yi-hack firmware and able to access SSH.
I wish I could use both app and yi-hack

roleoroleo, 2021-06-23:
> I just accidentally clicked "Reset to yi-hack default" 
> and then my camera cannot connect to app.

This is not a normal behavior.

> How could I reset and let the app able to scan wifi QR scan again?

Switch off the box "Disable cloud" and reset the cam using the button.

> I wish I could use both app and yi-hack
You can.

Switch off the box "Disable cloud".

vin-w, 2021-06-26:
```
reset the cam using the button.
```
@roleoroleo 
unfortunately the reset button is broken after opened the cam.
Possible to reset in other ways?

roleoroleo, 2021-06-26:
I don't know.

roleoroleo, 2021-07-03:
I tried to enable swap space but unfortunately the kernel doesn't support swap.
I can't add it.

News about the reset?

vin-w, 2021-07-04:
@roleoroleo 
Here is the memory when I have opened the hass page with webrtc + static image of cam periodically. Do you think its enough for the memory?

```
Load Average | 4.24 3.47 1.74
Free/Total Memory | 14708/60856 KB
Free SD Space | 9%
```

For reset, I need to solder the reset button again but no time yet. 
So I just leave the cam running.
Cannot get uptime more than 1 day, but in general it works most of the day with some reboot.

vin-w, 2021-07-04:
My guess on crash problem with hass: memory problem with too long imggrabber retrying?

I do another test and i suspect if concurrent access on imggrabber will have problem due to memory problem.
For example, if http call to snapshot was not completed, another one call http snapshot again, this will consume more memory.

What I did:
1. keep calling "imggrabber -r HIGH -w -d > /tmp/sd/snap.jpg"
2. when imggrabber is  keep printing "copy buffer again", call http snapshot
   http://<ip>/cgi-bin/snapshot.sh?res=high&watermark=yes
3. repeat 1+2 for many times
4. crashed

```
Free/Total Memory | 7256/60856 KB
```

```
/home/yi-hack/bin # imggrabber -r HIGH -w -d > /tmp/sd/snap.jpg
watermark on
debug on
Resolution: 1080
vaddr: 0xb4af8000 - paddr: 0x22fa0000 - size: 3502080
copy buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
copy again buffer: len 3502080
/home/yi-hack/bin # client_loop: send disconnect: Broken pipe
```

roleoroleo, 2021-07-04:
Yes, probably there is no memory for 2 sessions of imggrabber.
About webrtc audio, I fixed the problem. I will send a beta.

roleoroleo, 2021-07-04:
[h307_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6760355/h307_0.4.1.tar.gz)

vin-w, 2021-07-05:
> h307_0.4.1.tar.gz

@roleoroleo 
tried, but rtsp port 554 cannot connect.
Not sure why. Maybe rtsp crash?
Fallback firmware works again.

and after updating firmware, the cam always turn the cam up side top, and ptz cannot work again even I fall back firmware. (I disabled cloud) Hope I can fix reset button later but its hard to solder again.

roleoroleo, 2021-07-06:
I will check it.

roleoroleo, 2021-08-04:
Please try this:
[h307_0.4.1.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/6931388/h307_0.4.1.tar.gz)

vin-w, 2021-08-08:
@roleoroleo  no luck
 port 554: Connection refused

i tried both alaw/ulaw/disabled RTSP sound

roleoroleo, 2021-08-09:
Maybe...
Is rRTSPServer running?

vin-w, 2021-09-30:
@roleoroleo is there a version for h307_0.4.3.tar.gz ?
I dont find in here: https://github.com/roleoroleo/yi-hack-MStar/releases/tag/0.4.3

Thanks a lot!

roleoroleo, 2021-10-01:
Is still in beta, I haven't completed development.
Do you want to try it?

vin-w, 2021-10-01:
> Is still in beta, I haven't completed development. Do you want to try it?

@roleoroleo sure if I could help on the development

roleoroleo, 2021-10-01:
Here it is:
[h307_0.4.3.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7265716/h307_0.4.3.tar.gz)

vin-w, 2021-10-02:
> Here it is: [h307_0.4.3.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7265716/h307_0.4.3.tar.gz)

@roleoroleo it's working! Thanks a lot!

- RTSP video and audio OK in alaw setting in hass with webrtc
- ptz OK
- snapshot image OK

roleoroleo, 2021-10-02:
Ok  I will publish it with the next release.

gtramt1, 2021-12-06:
Hi, i tried first file uploaded, I flash it successfully (i think) because now it mostly speaks english. But after "wifi is connected" it turns up and starts speaking "this camera can only be used within china", i tried last uploaded firmware - same result. I tried reset - same thing...

Oh i am trying to connect via Yi home China, because if I use normal Yi home it asks to download china apk...

gtramt1, 2021-12-10:
Hi, i finallly succeded with flashing, but if i try to put wifi settings trough yi china app it turns away and probably start saying it can only work in china because it looks like speaker is burned (no sounds coming out) :D. it only connect if i put wifi settings via yi home global, but i dont get successful connection via app.

I can only access stream via tinycam pro...

any quick proxychains configuration manual maybe? or where to search it?

roleoroleo, 2021-12-10:
https://github.com/roleoroleo/yi-hack-MStar/wiki/Use-proxychains-ng-to-connect-your-app-(thanks-to-@x571)

gtramt1, 2021-12-11:
thanks, but it is impossible to get ssh. i tried root:empty, root: and set password, tried with custom username and password...always i get access denied (via putty)

roleoroleo, 2021-12-11:
If you don't change it in the configuration web page, the user is "root" and the password is blank.

roleoroleo, 2021-12-11:
The next release will contain a configuration page:
https://github.com/roleoroleo/yi-hack-MStar/commit/92899f999f289043476198d517ab8f0645297abb

gtramt1, 2021-12-11:
I did downgrade to '41, set proxychains...works everything except speaker... I disassembled camera but nothing looks suspicious...at least I have sound and video

kingtao1024, 2023-01-03:
> @roleoroleo I have dumped successfully
> 
> Here it is: LINK REMOVED
> 
> Thank you!

Hello friend, could you please send me a backup copy?
My camera model is the same as yours. I want an original firmware, but I can't get it from the customer service.
Thank you! My email: 254869033@qq.com

closed 2024-05-09 by github-actions[bot]