roleoroleo/yi-hack-MStar #370: Dome X Camera Will Not Turn On

Goobz1218, 2021-06-08:
Hi I'm not sure if you can help me but I received my Yi Dome X camera a couple of weeks ago and after running it for a week stock decided to flash your hack to it so I can integrate it with HA. It has worked flawlessly (besides the baby cry detection did nothing) until now but yesterday it died and now refuses to power up. When you plug the cord in nothing now happens.

Is there any advice you can give on getting it up and running again? Thanks for your help.

roleoroleo, 2021-06-08:
Maybe a hardware failure.
Does the led turn on?

Goobz1218, 2021-06-08:
Yeah that's what I feared. No it's completely dead. I'm guessing it doesn't leave me too many options then? I can try to return it but not sure if the fact it's hacked will affect that?

roleoroleo, 2021-06-09:
The hack can brick the cam but, in this case, you can recover it.
It's only a software brick.

Did you try to use another power adapter/cable?
Did you try to remove the sd card?

Goobz1218, 2021-06-09:
Yep tried all that. Still dead. Not sure what else I can try. I might try pulling it apart and see what I can find. Any clues on where to look?

I agree it looks more and more like a hardware failure. Thanks for your assistance and all your hard work!

closed 2021-06-09 by Goobz1218

roleoroleo, 2021-06-09:
If you have time to spend, connect a serial cable and check the log (it there is a log).

Goobz1218, 2021-06-11:
Thanks I'll give it a try when I get some time to have a look. Thanks for your help.