roleoroleo/yi-hack-MStar #372: RTSP sound issues

foxel, 2021-06-23:
I'm experiencing issues with sound over the RTSP stream. Started trying sound after I found https://github.com/AlexxIT/WebRTC/ which was promising without having the need for aac.

The camera I'm using: YI Home 1080

What I found is:
with alaw/ulaw  WebRTC sound sounds lower pitch.
VLC starts playing the sound and stops sound in 1-2 seconds.

The interesting thing though is that the newer Alwinner version of YI Home 1080 works OK with the same setup.

The difference I can see in VLC is that for the MStar camera it reports 16000 Hz sample rate, whereas for the Alwinner camera it reports 8000 Hz.

Any thoughts?

roleoroleo, 2021-06-23:
Which version?
Did you try 0.4.0? Before I added 2 way audio feature.

foxel, 2021-06-23:
MStar version 0.4.1 
Alwinner - yi-hack-Allwinner-v2 0.1.7

Didn't try 0.4.0. Is there an option to put a single alsa binary from 0.4.0 over 0.4.1 to check if it works?

roleoroleo, 2021-06-23:
I wi send it.

foxel, 2021-06-26:
So. I've tried 0.4.0 (downgraded the hack). Things stay the same.

vin-w, 2021-06-26:
I have the similar issue too @foxel  , to me alaw seems better

```
RTSP high (setting: ulaw audio)
VLC: stop audio after a few sec (the media info said it is mlaw, incorrect!)
HASS: webrtc integration: voice is distorted

RTSP high (setting: alaw audio)
VLC: clear voice
HASS: webrtc integration: voice is distorted
```

roleoroleo, 2021-06-26:
I noticed the same issue (I think it's the same): the sound is played slowly.
Probably webrtc doesn't understand 16 KHz alaw/ulaw audio.
I will try to convert to 8 KHz.

roleoroleo, 2021-07-02:
Changed to 8 KHz, it works.
I will fix in the next release.

Adjrobinson, 2021-09-01:
I would like to say this is amazing, the webrtc on my laptop allows me to have a babycam with crying alerts in a similar response / quality to the app. I am getting most success with ulaw. If I use home assistant ios app though at the same time it stops audio on the laptop and I never get audio in the ios app. Minor thing but I am not sure if there is a way around that at the moment, am I doing something wrong? Or would it be better easier an android phone. Thanks for this it is really good, it really has improved over the last year or so, I have now got another camera for our next baby!

roleoroleo, 2021-09-01:
> If I use home assistant ios app though at the same time it stops audio on the laptop and I never get audio in the ios app. Minor thing but I am not sure if there is a way around that at the moment, am I doing something wrong? 

Probably the cam doesn't have enough resources to send more than one stream.
If you want to use multiple clients, you need to install a server like motion to "republish" the stream.

Thanks for your donation!

closed 2021-09-24 by roleoroleo