roleoroleo/yi-hack-MStar #373: Unable to connect to web interface after a seemingly successful firmware flash

binarymale, 2021-06-29:
I have tried this on both my cameras, both are the Yi 1080p Home 4FUS model.
Firmware shown in Yi app: 4.5.0.0C_201910080934
I used: yi-hack-MStar 0.4.1 y203c

The following steps did happen correctly as far as I could see:
    The yellow light will come ON and flash for roughly 30 seconds, which means the firmware is being flashed successfully. The camera will boot up.
    The yellow light will come ON again for the final stage of flashing. This will take up to 2 minutes.
    Blue light should come ON indicating that your WiFi connection has been successful.

I can access the cameras within the Yi app, but unable to connect to the web interface using a laptop (different browsers tested), tried both the cameras IP address and the http://IP-CAM:8080 address.

Multiple power-cycles tried, re-copied flash files to SD card, but it just ignores them and camera boots quickly.
Also tried resetting camera and re-added to app.

As a side note, I did flash my Yi Outside camera using a different firmware that is compatible with that camera, and can access the web interface on that camera fine using the same laptop, I just don't know what I'm doing wrong here.

Thanks

roleoroleo, 2021-07-02:
Are you able to get the IP address of the cam?

binarymale, 2021-07-02:
Yes I know the IP addresses.
...And then I just tried again and it's working, I can now access the configuration of each camera.
Don't know what happened, all I did was put them back in their positions, rather than next to me in another room.

Anyway all is good, sorry to take up your time, and thanks for your work on this project :)

closed 2021-07-02 by binarymale