roleoroleo/yi-hack-MStar #376: CMSXJ16A support

gxcreator, 2021-07-29:
**CMSXJ16A** aka Mijia Xiaobai IMILAB Home Security Camera Basic

[image]
[image]
[image]
[image]

**UART**
Connect to pads as shown at picture above
**uBoot**
Keep pressing Enter during cold start,
```IPL gd156225
D-01.
HW Reset
64MB
BIST0_0001-OK
offset:00010000
size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4
MXP found at 0x00020000
  decomp_size=0x0004ad64

U-Boot 2015.01 (Aug 12 2019 - 13:56:26), Build: jenkins-ipc016_revert_tutk-2

Version: I3gfe5f65a
DEVINFO: 313E
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
gpio[100] is 1
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x090F, 0x1C, 0x70, 0x18)
SF: Detected nor0 with total size 16 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   MAC Address 00:30:1B:BA:02:DB
Auto-Negotiation...
AN failLink Status Speed:10 Full-duplex:0
Status Error!
mstar_emac
Warning: mstar_emac using MAC address from net device

MStar #
MStar #
```

**Layout**
```
MStar # printenv
baudrate=115200
bootcmd=sf probe 0;sf read 0x21000000 0x50000 0x260000;bootm 0x21000000
bootdelay=0
dbgLevel=DEBUG

```

```
mtd0: 00050000 00010000 "BOOT"
mtd1: 00210000 00010000 "KERNEL"
mtd2: 00750000 00010000 "ROOTFS"
mtd3: 00630000 00010000 "DATA"
mtd4: 00010000 00010000 "CONFIG"
mtd5: 00010000 00010000 "FACTORY"
```

```
MStar # mxp t.list
Total MXP record count:12
[mxp_record]: 0
     name: IPL
     type: 0x01
   format: 0x00
   backup:
    start: 0x00004000
     size: 0x0000C000
   status: 0x00

[mxp_record]: 1
     name: IPL_CUST
     type: 0x01
   format: 0x00
   backup:
    start: 0x00010000
     size: 0x0000F000
   status: 0x00

[mxp_record]: 2
     name: KEY_CUST
     type: 0x01
   format: 0x00
   backup:
    start: 0x0001F000
     size: 0x00001000
   status: 0x00

[mxp_record]: 3
     name: MXPT
     type: 0x01
   format: 0x00
   backup:
    start: 0x00020000
     size: 0x00001000
   status: 0x00

[mxp_record]: 4
     name: UBOOT
     type: 0x01
   format: 0x00
   backup:
    start: 0x00030000
     size: 0x0001F000
   status: 0x00

[mxp_record]: 5
     name: UBOOT_ENV
     type: 0x01
   format: 0x00
   backup:
    start: 0x0004F000
     size: 0x00001000
   status: 0x00

[mxp_record]: 6
     name: BOOT
     type: 0x03
   format: 0x00
   backup:
    start: 0x00000000
     size: 0x00050000
   status: 0x00

[mxp_record]: 7
     name: KERNEL
     type: 0x03
   format: 0x00
   backup:
    start: 0x00050000
     size: 0x00210000
   status: 0x00

[mxp_record]: 8
     name: ROOTFS
     type: 0x03
   format: 0x00
   backup:
    start: 0x00260000
     size: 0x00750000
   status: 0x00

[mxp_record]: 9
     name: DATA
     type: 0x03
   format: 0x00
   backup:
    start: 0x009B0000
     size: 0x00630000
   status: 0x00

[mxp_record]: 10
     name: CONFIG
     type: 0x03
   format: 0x00
   backup:
    start: 0x00FE0000
     size: 0x00010000
   status: 0x00

[mxp_record]: 11
     name: FACTORY
     type: 0x07
   format: 0x00
   backup:
    start: 0x00FF0000
     size: 0x00010000
   status: 0x00

```

**Linux**
Just boot without enttering u-boot. User: root with no password

**Connecting to wifi and enabling ssh**
Get ssh running, replace http://192.168.1.224:8000/hacks/ssh-server/bin/dropbear with your url to armv7l dropbear
```
killall hostapd
echo -ne "network={\n    ssid=\"gx-iot\"\n    psk=\"YOUR_PASSWORD\"\n}\n" > /tmp/wpa.conf
wpa_supplicant -B -i wlan0 -c /tmp/wpa.conf -D wext

udhcpc -i wlan0

ping -c 2 8.8.8.8

curl -k http://192.168.1.224:8000/hacks/ssh-server/bin/dropbear --output /tmp/dropbear
cd /tmp
chmod +x dropbear

# cat config.json
echo -ne "{\"users\": [{\"systemUsername\": \"root\", \"username\": \"root\", \"password\": \"\"}]}" > config.json
./dropbear -FREB -r ./host_150601_rsa -r ./host_150601_rsa -r ./host_150601_rsa -C config.json
```

**Dumping FS**
```
dd if=/dev/mtd0 of=/tmp/mtd0.bin
dd if=/dev/mtd1 of=/tmp/mtd1.bin
dd if=/dev/mtd2 of=/tmp/mtd2.bin
dd if=/dev/mtd3 of=/tmp/mtd3.bin
dd if=/dev/mtd4 of=/tmp/mtd4.bin
dd if=/dev/mtd5 of=/tmp/mtd5.bin
```
Then connect to ssh and grab this files.

gxcreator, 2021-07-29:
[Mijia_cam-mtdblocks-dump.zip](https://github.com/roleoroleo/yi-hack-MStar/files/6903727/Mijia_cam-mtdblocks-dump.zip)
Partitions dump.

gxcreator, 2021-07-29:
Could someone guide me if I could use scripts from this repo as-is or need to tweak something specific for this camera?

roleoroleo, 2021-08-04:
> Could someone guide me if I could use scripts from this repo as-is or need to tweak something specific for this camera?

The chipset is the same but the system is completely different.
You can't use the script from this repo.

ivan-leschinsky, 2021-08-04:
@gxcreator I think we can try to run `framegrabber` directly after ssh to the camera, As I understood it grabs frames from camera and accessible cia some protocols(RTSP maybe). https://github.com/cmiguelcabral/mjsxj05cm-hacks/blob/dev/sdcard/hacks/framegrabber/bin/framegrabber 
I also wanted to check this but I still didn't opened camera as I'm using it twice a day

roleoroleo, 2021-08-04:
The libc is also different.

closed 2021-09-24 by roleoroleo

evilmumi, 2023-11-25:
any1 with custom firmware for that cam?

floxigen, 2024-03-03:
This camera is Mstar can maybe this hack work ?