roleoroleo/yi-hack-MStar #380: Reboot Loop  (h201c - 0.4.1)

bepece1, 2021-08-21:
I have a issue with my cam (YiDomeCamera1080p) so as my cam always is reboting and the native app no work
I have tried downgrade hack version to 0.3.5 but the same problem go on in my cam.
What can I do? some ideas?

Yihack always has been working in this cam, I don't know what happen now

YiHack Version  0.4.1 (0.3.5)
Base Version 4.6.0.0A_201908271549
Model  h201c
[log.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7025301/log.zip)

thanks in advance

roleoroleo, 2021-08-23:
If you change the fw version (i.e. 0.4.1 -> 0.3.5 or another versione) the problem should disappear because the cam is completely rewritten.
Maybe the problem is related to a power failure.
Did you change the power adapter or the power cable?

roleoroleo, 2021-08-23:
The log shows no problems.

bepece1, 2021-08-24:
> If you change the fw version (i.e. 0.4.1 -> 0.3.5 or another versione) the problem should disappear because the cam is completely rewritten.
> Maybe the problem is related to a power failure.
> Did you change the power adapter or the power cable?

Yes, I tried different power adapters with other wires and the result is the same 😣

How can I recopile a avanced log?

Exists other way to reset fully the cam?

roleoroleo, 2021-08-24:
> How can I recopile a avanced log?

Is not possible.

> Exists other way to reset fully the cam?

Restore the original partitions.
But IMHO it will not fix the problem.

bepece1, 2021-08-25:
> > How can I recopile a avanced log?
> 
> Is not possible.
> 
> > Exists other way to reset fully the cam?
> 
> Restore the original partitions and reset the cam with the back button.
> But IMHO it will not fix the problem.

How would you fix this problem?

roleoroleo, 2021-08-25:
I don't know.
Maybe a hardware failure.

But try to restore the original partition.

bepece1, 2021-08-25:
> But try to restore the original partition.

Can you send me a link with information how do it?

Thanks in advance

bepece1, 2021-09-02:
I go on with the problem.

If I execute "dmesg" and I can see some errors, do you know what it can happen?

/home/yi-hack # dmesg
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0x21 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0x21 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0x21 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0x21 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x0 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x0 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x0 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x0 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x0 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x6 0x1
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x6 0x1
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x6 0x1
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x6 0x1
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x3e 0x6 0x1
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0xe 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0xe 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0xe 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0xe 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x32 0xe 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x38 0x12 0x0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x1 0xd0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x1 0xd0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x1 0xd0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x1 0xd0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x1 0xd0
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x14 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x14 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x14 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x14 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x33 0x14 0x4
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x57 0x81 0x2
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x57 0x81 0x2
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x57 0x81 0x2
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x57 0x81 0x2
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x57 0x81 0x2
ERROR: Bus[1] in ms_i2c_xfer_write: Slave dev NAK, Addr: 0x60, Data: 0x38 0x12 0x30
[PWN] mstar_pwm_config duty_ns=100, period_ns=100000
reg=0x1F003490 clk=12000000, period=0x78
reg=0x1F003488 clk=12000000, u32Duty=0x78
[PWM] mstar_pwm_enable
[SCLIRQ]chang time :26354

bepece1, 2021-10-04:
I think I have a space problem in root, how can I free space in root? @roleoroleo 

/ # df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 1920      1904        16  99% /
tmpfs                    30428       128     30300   0% /dev
/dev/block/mtdblock3     12480     11252      1228  90% /home
tmpfs                    32768        76     32692   0% /tmp
/dev/block/mmcblk0p1   1951676     12636   1939040   1% /tmp/sd

andrzejSlomiany, 2021-10-05:
I have the same problem with the Yi Dome Camera 1080p.
Camera keeps resetting after rebooting and setting up wifi connection.
YiHack Version 0.4.3
Base Version 4.6.0.0A_201908271549
Model h201c
Please help
[Logi.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7288597/Logi.zip)

andrzejSlomiany, 2021-10-08:
After typing the command ls -l /etc/resolv.conf gets:
/tmp$ ls -l /etc/resolv.conf 
lrwxrwxrwx    1 root     0               16 Oct  8 18:54 /etc/resolv.conf -> /tmp/resolv.conf
What the /tmp/resolv.conf file should contain.
There is in my file:
nameserver 192.168.100.1
nameserver 8.8.8.8

closed 2024-05-08 by github-actions[bot]