roleoroleo/yi-hack-MStar #381: Crontab does not save the command

NikDevx, 2021-08-30:
Command saved success
[image]

But after rebooting I see this
[image]

roleoroleo, 2021-08-30:
It should work:
https://github.com/roleoroleo/yi-hack-MStar/commit/c5ba66ba89983ae7a5dab40fd65788772448552e
Try 0.4.2.

NikDevx, 2021-08-30:
> It should work:
> [c5ba66b](https://github.com/roleoroleo/yi-hack-MStar/commit/c5ba66ba89983ae7a5dab40fd65788772448552e)
> Try 0.4.2.

Thanks. That helped!

NikDevx, 2021-09-01:
Hello
0.4.2 and 0.4.3 not working `speaker.sh`

```
{
"error":"false"
"description":""
}
```

roleoroleo, 2021-09-01:
Please open a new issue if not related to crontab.

closed 2021-09-01 by NikDevx