roleoroleo/yi-hack-MStar #382: RTSP stream issues after upgrade to 0.4.2

pejotigrek, 2021-08-30:
camera model:
- y203c
- h201c

everything worked fine on 0.4.1 but after I upgraded today to latest 0.4.2 [using OTA]  RTSP stream broke. neither VLC, Frigate NVR [which I use as motion/object detection in HomeAssistant] or iSpy [on Windows] can connect to the camera - and they all could before. WWW dashboard & snapshots work fine.

Madaraxa, 2021-08-30:
Same here. The Allwinner Update doesnt have this problem, if this info helps at all

roleoroleo, 2021-08-30:
Wow :(
Please, help me, I need to debug the problem because in my y203c is working properly.
Are you able to check if the process is running?
What's your system config?

pejotigrek, 2021-08-30:
@roleoroleo can you guide me a litle bit - I guess I should check the process after ssh to camera, right?

what system's config I should provide?

(forgive my those questions, I'm not too fluent in linux based environments 😅)

roleoroleo, 2021-08-30:
Login with ssh and run the command `ps`
Post the content of the file /home/yi-hack/etc/system.conf
Hide your personal data.

Madaraxa, 2021-08-30:
Hi. I did this by using 'cd etc' and 'vi system.conf' i am really bad at using Linux ;) 
`
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=no
RTSP_AUDIO_NR_LEVEL=0
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=low
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=root
PASSWORD=mypw
FREE_SPACE=50
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_USERNAME=
FTP_PASSWORD=
SSH_PASSWORD=mypw`

roleoroleo, 2021-08-30:
Please, try to overwrite rRTSPServer with this one:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-MStar/files/7078906/rRTSPServer.gz)

Unzip it in /home/yi-hack/bin and check that permissions are 755.

roleoroleo, 2021-08-30:
If it works I will publish a new release.

Madaraxa, 2021-08-30:
Man you are incredible. This one works :) i just needed to get some Infos of Google how to unzip etc :)

roleoroleo, 2021-08-30:
Please try 0.4.3.

Madaraxa, 2021-08-30:
It works :)

pejotigrek, 2021-08-30:
@roleoroleo thanks for the hints and forgive my delay [family duties called ;)]
for the record - here are my outputs: 
[y203c_ps_output.txt](https://github.com/roleoroleo/yi-hack-MStar/files/7079291/y203c_ps_output.txt)
[y203c_system_conf.txt](https://github.com/roleoroleo/yi-hack-MStar/files/7079289/y203c_system_conf.txt)
[h201c_ps_output.txt](https://github.com/roleoroleo/yi-hack-MStar/files/7079290/h201c_ps_output.txt)
[h201c_system_conf.txt](https://github.com/roleoroleo/yi-hack-MStar/files/7079292/h201c_system_conf.txt)

also - tried 0.4.3 and it works great on both my cameras. you're the man!! thank you for such a quick response & fix!

closed 2021-08-30 by pejotigrek