roleoroleo/yi-hack-MStar #386: Is there an Android app?

giejay, 2021-09-08:
Thanks for this awesome "hack"! I'm planning on building an Android app because the Yi Home app does not allow me to retrieve motion videos very easily (afaik) from sd-card/ftp server. Only when paying for the cloud. 

Tinycam is a really good app but also does not show me a nice timeline of videos stored on my ftp server (only locally recorded videos using Tinycam).

So, what I would like is an Android app which can play the RTSP stream including audio, and show a timeline of all captured videos on my ftp server below the video. Any idea if this is already built by someone before I start spending time on this?

roleoroleo, 2021-09-10:
I've never seen an app like this.
I'm using Tinycam and Onvifer.

giejay, 2021-09-10:
I started building this but the main problem I'm facing is the instability of the stream when audio is included. It crashes or a lot of artifacts appear. 

I'm wondering how the native app can work so smoothly including audio, I'm guessing it uses a different streaming method/codec?

Are there supported cameras with a bit more CPU/memory?

roleoroleo, 2021-09-11:
> I'm wondering how the native app can work so smoothly including audio, I'm guessing it uses a different streaming method/codec?

Different method, yes. The codec is the same: h264.

> Are there supported cameras with a bit more CPU/memory?

Supported by this hack, no.

giejay, 2021-09-11:
Is it possible to offload the streaming process to a server (raspberry pi) like it's done in this project for Wyze cams: https://github.com/mrlt8/docker-wyze-bridge?

roleoroleo, 2021-09-11:
I don't know.
But it's not normal this behavior. My stream is not unstable.

vin-w, 2021-09-30:
> Is it possible to offload the streaming process to a server (raspberry pi) like it's done in this project for Wyze cams: https://github.com/mrlt8/docker-wyze-bridge?

I have thought of this idea too.
The problem now I have is in home assistant, there are multiple user viewing the stream.
It will crash the camera.

Running a rtsp proxy before the camera stream would help.
But my experience with yi-cam x rtsp-simple-server was not good.
Will try another streaming proxy later

Rocket200, 2021-10-02:
i use
ipcamviewer in  ios android and firetv
instarvision with generiv h.264 profil for yi cam

iocamviewer on firetv :) the best now 6 Cams (5xyi,1xeufy)
what audio is the best ? pcm ?

eufy work without internet !!
but only can view the rtsp stream with sound!

roleoroleo, 2021-10-02:
If you mean for the quality, the best is pcm.
But in general alaw or ulaw are more compatible and generate less traffic.

closed 2024-05-08 by github-actions[bot]