roleoroleo/yi-hack-MStar #389: ONVIF control via zoneminder broken after upgrading from 0.4.1 to 0.4.3

ronaldlw, 2021-09-20:
After upgrading I am no longer able to control the camera PTZ via zoneminder.  I use the following settings in the monitor for the camera:
controllable (checked)
control type: ONVIF Camera
control device: empty
control address: <camera IP>:80
the rest is empty, unchecked or NONE

roleoroleo, 2021-09-20:
Please check the options used to start onvif daemon:
```
cat /tmp/onvif_srvd.conf
```

ronaldlw, 2021-09-20:
I don't see that file:

/home/yi-hack # ls -1 /tmp
MTK
audio
audio_fifo
audio_in_fifo
debug_alarm.txt
debug_oss.txt
debug_p2p.txt
dns1
done
gw1
h264_high_fifo
h264_low_fifo
hosts
httpd.conf
log.txt
log_oss.txt
logsock
mask
mmap.info
resolv.conf
sd
var
wpa_supplicant.conf

ronaldlw, 2021-09-20:
After upgrading to 0.4.3. again:
/home/yi-hack # cat /tmp/onvif_srvd.conf
pid_file=/var/run/onvif_srvd.pid
model=Yi Hack
manufacturer=Yi
firmware_ver=0.4.3
hardware_id=BFUS
serial_num=BFUSY1E14YJM07191126
ifs=wlan0
port=80
scope=onvif://www.onvif.org/Profile/S

#Profile 0
name=Profile_0
width=1920
height=1080
url=rtsp://%s/ch0_0.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=high
type=H264

#PTZ
ptz=1
move_left=/tmp/sd/yi-hack/bin/ipc_cmd -m left
move_right=/tmp/sd/yi-hack/bin/ipc_cmd -m right
move_up=/tmp/sd/yi-hack/bin/ipc_cmd -m up
move_down=/tmp/sd/yi-hack/bin/ipc_cmd -m down
move_stop=/tmp/sd/yi-hack/bin/ipc_cmd -m stop
move_preset=/tmp/sd/yi-hack/bin/ipc_cmd -p %t

roleoroleo, 2021-09-20:
The file is ok.
Is the daemon running?
Check with "ps".

ronaldlw, 2021-09-20:
yep 
 1151 root      0:00 onvif_srvd --conf_file /tmp/onvif_srvd.conf

roleoroleo, 2021-09-21:
The process is ok and the configuration file is ok.
Please, test PTZ using "Onvif Device Manager".
Are you using authentication?

ronaldlw, 2021-09-22:
What is "onvif device manager"?

I am not using auth

roleoroleo, 2021-09-22:
It's a windows softwtare useful to test the onvif protocol.
https://sourceforge.net/projects/onvifdm/

ronaldlw, 2021-09-22:
ptz does not work using that tool either.

ronaldlw, 2021-09-22:
Firmware Version | 0.4.3
Base Version | 4.6.0.0A_201908271549
Model Suffix | h201c
Serial Number | BFUSY....

roleoroleo, 2021-09-22:
Ok, wrong path.
Check this commit: https://github.com/roleoroleo/yi-hack-MStar/commit/b0e69281889746ba3b01607daf37e6e24481455f
You can edit files yourself.

ronaldlw, 2021-09-22:
is a reboot required after editing the files?

ronaldlw, 2021-09-22:
ok, i went ahead and rebooted and it's working now.  To be clear, I updated these files with the changes from your commit:
/home/yi-hack/script/system.sh
/home/yi-hack/www/cgi-bin/service.sh

closed 2021-09-24 by roleoroleo