roleoroleo/yi-hack-MStar #390: Completely dead BFUS black dome camera

Kaan88, 2021-09-23:
Hi, camera was acting weird and then all of a sudden it died. There is no light or any movement, just some hiss from the speaker.

I tried serial connection with both my Pi zero and ch341 flash reader. (pi worked fine with my 1s camera) There is no bootloader activity at all.

At this point I assume it's a hardware issue and not a software brick. Since there was no backup I couldn't restore anything anyway. I'm going to attach the backup I took after the hardware died, maybe it'll shed some light? 
[dead_bfus.zip](https://github.com/roleoroleo/yi-hack-MStar/files/7217828/dead_bfus.zip)

Ps. By acting weird I mean, for example, even though "disable cloud" was disabled, I couldn't connec to Yi App all day, and the videos were glitchy.

roleoroleo, 2021-09-24:
> At this point I assume it's a hardware issue and not a software brick. Since there was no backup I couldn't restore anything anyway. I'm going to attach the backup I took after the hardware died, maybe it'll shed some light?

Your backup seems ok.
uboot and kernel are ok.
And I'm able to mount rootfs and home partitions without errors.

Kaan88, 2021-09-24:
Thanks, it's hardware then.

closed 2021-09-24 by Kaan88