roleoroleo/yi-hack-MStar #391: Can't connect to WiFi after lost connection.

Chreece, 2021-10-05:
Hello, 
Thank you very much for your efforts, on the firmware. It's really a all in one solution!
I'm experiencing a problem. If the camera loses the connection to WiFi (i.e. after reboot or power off) it needs a re authentication and connection to yihome app to retrieve the WiFi settings. All saved settings are there after that, but somewhat after a lost connection the camera can't connect again to WiFi. 
Note: I disabled app-mode (private connection)

roleoroleo, 2021-10-06:
I've seen this behavior before but have never been able to understand why it happens and find a solution.

roleoroleo, 2021-10-07:
> Are there any logs that I can send you?

No.
Probably it's a race condition, but with my cam it never happened.

closed 2024-05-08 by github-actions[bot]